sparkmicro/Ki-nTree #158: How to fix "Warning: Parameter map for "libraryXYZ" does not exist or is empty"

T0jan, 2023-05-22:
Hi,

I am struggling to get the pulled parameters from Ki-nTree into my KiCad libraries. The parameter fields from the template just keep their placeholder and Ki-nTree throws following warning if started in a console:
```
[INFO]	Warning: Parameter map for "libraryXYZ" does not exist or is empty
```
I suspect one has to add the mapping into the KICAD_LIBRARIES section of [kicad_map.yaml](https://github.com/sparkmicro/Ki-nTree/blob/main/kintree/config/kicad/kicad_map.yaml) but without any examples or documentation it is not clear what to add how there.

eeintech, 2023-06-04:
Hello @T0jan

It is not documented unfortunately but it comes from [kintree/config/inventree/supplier_parameters.yaml](https://github.com/sparkmicro/Ki-nTree/blob/322f40f38296dc9795ff4a349bb65721b4713c8f/kintree/config/inventree/supplier_parameters.yaml) file, you need to setup a mapping between the supplier parameters linked to the part you're adding and the InvenTree parameters.

If you haven't already, you can define InvenTree parameters in this file: https://github.com/sparkmicro/Ki-nTree/blob/322f40f38296dc9795ff4a349bb65721b4713c8f/kintree/config/inventree/parameters.yaml

And run the `kintree_setup_inventree` script to create them in InvenTree.

I hope it helps!

T0jan, 2023-06-21:
Actually it was much easier then expected. you don't need to touch `kicad_map.yaml` at all. in fact you don't need to do anything for kicad except selecting the library and template you want in the UI. everything that is needed, is that the `Categories` in `supplier_parameters.yaml` match the categories in InvenTree. of course the parameter names also need to match the parameters in InvenTree so if you have custom categories and parameters like me this obviously needs to be adjusted. as you said, if you run `kintree_setup_inventree` before you start to fill the database this is not a problem.

closed 2023-06-21 by T0jan

eeintech, 2023-06-22:
Glad you were able to make this work!