sparkmicro/Ki-nTree #159: Crash when creating part in Inventree

SparkeyinVA, 2023-05-30:
I have recently upgraded to Ki-nTree 1.0.1, Kicad 7.0, and Inventree 0.11.2.
- After the update, I have verified that all connections between the parts supplier and Ki-nTree are working. The connection between Ki-nTree and Inventree is correct, and the connection is with a token, and the connection to Kicad is correct.
- before update I could make a part in Inventree but not Kicad.
- After the update I can make the part in Kicad correctly, but i get a crash when making the part in Inventree. 
- Here is the information shown in PowerShell. I am not sure what is missing or messed up.  Any pointers is great.

[MAIN]  Digi-Key search for 311-16.2KLRct-nd
[TREE]  Successfully connected to InvenTree server (ENV=DEVELOPMENT)
[INFO]  The following parameters are not mapped in Digi-Key parameters configuration:
['Packaging', 'Features', 'Package / Case', 'Temperature Coefficient', 'Size / Dimension', 'Composition', 'Number of Terminations', 'Height - Seated (Max)', 'Failure Rate', 'Part Status']
Exception in thread Thread-158 (create_part):
Traceback (most recent call last):
  File "C:\Users\jchamberlain\AppData\Local\Programs\Python\Python310\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\Users\jchamberlain\AppData\Local\Programs\Python\Python310\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\jchamberlain\AppData\Local\Programs\Python\Python310\lib\site-packages\kintree\gui\views\main.py", line 1196, in create_part
    new_part, part_pk, part_info = inventree_interface.inventree_create(
  File "C:\Users\jchamberlain\AppData\Local\Programs\Python\Python310\lib\site-packages\kintree\database\inventree_interface.py", line 525, in inventree_create
    part_pk = inventree_api.create_part(category_id=category_pk,
  File "C:\Users\jchamberlain\AppData\Local\Programs\Python\Python310\lib\site-packages\kintree\database\inventree_api.py", line 357, in create_part
    part = Part.create(inventree_api, {
  File "C:\Users\jchamberlain\AppData\Local\Programs\Python\Python310\lib\site-packages\inventree\base.py", line 144, in create
    response = api.post(cls.URL, data, **kwargs)
  File "C:\Users\jchamberlain\AppData\Local\Programs\Python\Python310\lib\site-packages\inventree\api.py", line 392, in post
    response = self.request(
  File "C:\Users\jchamberlain\AppData\Local\Programs\Python\Python310\lib\site-packages\inventree\api.py", line 344, in request
    raise requests.exceptions.HTTPError(detail)
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url': 'http://192.168.1.200/api/part/', 'method': 'POST', 'status_code': 500, 'body': '{"error":"TypeError","error_class":"<class \'TypeError\'>","detail":"Error details can be found in the admin panel","path":"/api/part/","status_code":500}', 'headers': {'AUTHORIZATION': 'Token 506e0ca133cae351c96925e1a37b35e6e207eac1'}, 'params': {'format': 'json'}, 'data': {'name': 'RES 16.2K OHM 1% 1/16W 0402', 'description': 'RES 16.2K OHM 1% 1/16W 0402', 'category': 235, 'keywords': 'RES 16.2K OHM 1% 1/16W 0402', 'revision': 'A', 'active': True, 'virtual': False, 'component': True, 'purchaseable': True}}

eeintech, 2023-06-04:
Hello @SparkeyinVA 

It looks like a very generic API error... it could be that the way Ki-nTree creates part in InvenTree doesn't handle your example.
Have you succeeded to create any other part on this setup?
Have you looked in the admin section of InvenTree for more details?
`"detail":"Error details can be found in the admin panel"`

eeintech, 2023-06-04:
Also I have never tried to run Ki-nTree in powershell... I'm a little concerned about it 😕

eeintech, 2023-08-03:
Hello @SparkeyinVA Is this issue this pending resolution on your side?

eeintech, 2023-08-04:
I cannot reproduce this issue using [Git Bash](https://git-scm.com/downloads):

``` shell
$ python -m kintree_gui

[MAIN]  Digi-Key search for 311-16.2KLRct-nd
[TREE]  Successfully connected to InvenTree server (ENV=DEVELOPMENT)
[INFO]  The following parameters are not mapped in Digi-Key parameters configuration:
['Packaging', 'Features', 'Package / Case', 'Temperature Coefficient', 'Size / Dimension', 'Composition', 'Number of Terminations', 'Height - Seated (Max)', 'Failure Rate']

[MAIN]  Generating Internal Part Number
[INFO]  Internal Part Number = RES-000005-00
[INFO]  Success: Added new part to InvenTree
```

Please switch to Git Bash in order to run Ki-nTree.

I will close this issue for now, feel free to reopen it if you still encounter the same crash in Git Bash.

closed 2023-08-04 by eeintech