sparkmicro/Ki-nTree #163: [bug] crash when using alternate mode with IPN, parameter do not get added

T0jan, 2023-06-21:
Hi,

I tried to update existing parts with Ki-nTree today but run into two problems:

* When entering a IPN and no Part ID as suggested by the tool tip the request to inventree results in a timeout. 
* Using the ID finds the part but does not add any new parameters to it or even checks them. Manufacturer Part and Supplier part however are checked. Not sure if this is intended?

Greetings.

T0jan, 2023-06-22:
source for the first bug is this line here:

https://github.com/sparkmicro/Ki-nTree/blob/322f40f38296dc9795ff4a349bb65721b4713c8f/kintree/database/inventree_api.py#L171

instead of downloading just the entry for the given IPN it tries to download the whole database which obviously will end in a timeout for bigger databases. Changing the line to `parts = Parts.list(inventree_api, IPN=part_ipn)` or `parts = Parts.list(inventree_api, IPN_regex=part_ipn)` in case a more lenient search is intended should fix this. also would mean, that the loop afterwards could be replaced by a simple if condition.

eeintech, 2023-06-22:
Hi @T0jan, good catch!

Do you want to send a PR for this?

T0jan, 2023-06-22:
@eeintech I will fix it as part of the implementation of #162. However any opinions on the other point? 

> Using the ID finds the part but does not add any new parameters to it or even checks them. Manufacturer Part and Supplier part however are checked. Not sure if this is intended?

eeintech, 2023-06-22:
> @eeintech I will fix it as part of the implementation of #162. However any opinions on the other point?
> 
> > Using the ID finds the part but does not add any new parameters to it or even checks them. Manufacturer Part and Supplier part however are checked. Not sure if this is intended?

I actually didn't intend the Part ID / IPN to be used to **_update_** parts but only to add alternates. This is an interesting use case though... In this case the "Alternate" switch should be more generic, but I'm not sure how I feel about it yet.
Also have you tried to use the manufacturer PN to do a new search and add from scratch? It should find your already existing part but can't remember if parameters are updated... will have to try later

eeintech, 2023-06-22:
@T0jan Umm I'm surprised this is not obvious from UX standpoint:

[image]

What does "Alternate" means to you?

T0jan, 2023-06-22:
@eeintech I understood it as an alternate input mask to the standard category/IPN Category Code selection. But with your explanation I see your point. maybe instead of modifying the alternate mode adding a second switch for an similar update mode would be an idea?

eeintech, 2023-06-22:
Oh I see, this name is so telling to me in the world of electronic components that I forgot the generic meaning can be lead to other interpretation. Maybe an "Create/Update" switch would be a good idea 👍

T0jan, 2023-08-04:
fixed with #167, closing

closed 2023-08-04 by T0jan