sparkmicro/Ki-nTree #165: [bug] InvenTree 0.12.0 broke parameter handling

T0jan, 2023-07-03:
The recent changes to InvenTree parameter handling especially the now way harsher filtering of parameter values breaks the current approach of just taking the values from e.g. digikey and creating new part parameters with them. As of now no special characters are allowed for parameters with standard units and also empty parameter with `-` as placeholder are no longer possible. I opened a discussion on [IvenTree](https://github.com/inventree/InvenTree/discussions/5140) for them to review the changes but maybe there also needs to be developed a workaround in the meantime.

eeintech, 2023-07-04:
@T0jan Would https://github.com/inventree/InvenTree/pull/5160 fix this problem?

T0jan, 2023-07-04:
@eeintech probably, as it resulted from the discussion, have not tested yet.

T0jan, 2023-07-21:
So after a bit of testing I can confirm it works. As they have put this fix behind a server setting now we should add a note to the README to make sure everyone who uses Ki-nTree knows he has to set this flag.

I also thought a bit if there is a way to automatically convert the non compliant units the suppliers use (e.g. "Ohms" instead of "ohm"), but the only usable idea I got would be look-up tables and I don't want to go down this road in this case here.

eeintech, 2023-07-26:
Hello @T0jan Thanks for the feedback, do you know why this fix wasn't brought in into point releases of v12?

T0jan, 2023-07-26:
@eeintech as far as I understand they don't really see it as bug/fix but more as a feature added to the new unit control they added with 0.12.

kongr45gpen, 2023-09-07:
https://github.com/inventree/InvenTree/pull/5160 was merged in [Inventree 0.12.6](https://github.com/inventree/InvenTree/releases/tag/0.12.6).

I have tested Ki-nTree 0.5.3 and it works fine on my Inventree 0.12.6 with a sample DigiKey component (and parameters seem to be transferred fine).

eeintech, 2023-09-07:
Hi @kongr45gpen Yes @T0jan also confirmed it works with the latest Ki-nTree version too. However for `0.13.x` all InvenTree users will need to toggle a setting to enable compatibility, we will document it.

closed 2023-10-14 by eeintech

T0jan, 2023-11-06:
Small note here: InvenTree removed the enforced checks on parameters completely for now so it is not needed to set the switch in the settings (the switch actually is not there anymore from at least 0.12.8 onward.

eeintech, 2023-11-06:
@T0jan That's good news, less doc for us!

eeintech, 2023-12-22:
@T0jan Good point. I haven't had much time those last few weeks/months but I'm going to try to push a new release early next year.

PR #180 (2023-10-14): 1.0.4
* #172: Add supports for template handling for InvenTree version>=0.12.6 and fixes #165
* #173: Add price sanitization
* #175: Mouser parts use MPN for part name instead of description
* #179: Check that pricing data upload option is enabled by user
* Fix #171 bug with handling empty category `CODES`