sparkmicro/Ki-nTree #168: [Feature Request] QR code scanner

coratron, 2023-07-20:
Salut François!

I have been looking into solutions to manage my inventory (of electronic components) and I stumbled upon Inventree and Ki-nTree.

It looks like Ki-nTree is mostly targeting new projects (when you are hunting for parts - a priori), so this feature might be slightly out of scope in that regard.

I have reels and bags of components here, and need to enter them into the database (a posteriori approach). I came across some other projects such as:

- https://github.com/maholli/getparts

which has a fork that looks interesting:

- https://github.com/Decee1/SupplierBarcodeAnalyzer

I have been successfully scanning items here, but that's where my exploration ends. I looked into custom plugins for Inventree but it looks like Ki-nTree has all the features.

You are probably aware of these solutions and I was wondering if you have considered integrating a QR code scanner into Ki-nTree. I am hoping this would not be a huge rework as the scanner would just look for the part number to do a full search and retrieval of data via your system. Instead of a textfield with the part number, you would have the option to open the webcam (using scripts like Decce1's) which returns a json with all the info.

The only thing that is perhaps not part of Ki-nTree is entering the amount of components which is also available on the QR code, but manually entering that is not a huge problem once the component is in Inventree's DB?

What do you think?

A+

eeintech, 2023-07-26:
Hello @coratron 

If I'm understanding correctly, you'd like to scan bar codes to create InvenTree stock items of already existing part in your database? Isn't it already a feature of InvenTree mobile app? I thought it was but I don't use it 🤔 

The purpose of Ki-nTree is to create parts from scratch like you said, but not about managing stock items, this is outside the scope of this project unfortunately.

coratron, 2023-08-02:
The items are not in the database yet. And it seems the Inventree QR code system only works for items you have already created and generated a QR code for. The general flow here would be:
1- Buy and item from Digikey, LCSC, Mouser, etc
2- Scan the QR code via Ki-nTree 
3- Ki-nTree generates an entry in the Inventree database with all the right parameters for the electronic components

I switched to KiCAD recently so I have all these items around, which is why this is my approach instead of starting from scratch and picking components. I already have preferred parts in stock.

eeintech, 2023-08-03:
Ok now I understand what you are trying to do. Unfortunately I don't have much experience in barcode scanning but I'm sure it is something doable. My time now on Ki-nTree is _very_ limited to say the least... do you have time to take a stab at it?

T0jan, 2023-08-22:
This is an interesting idea but would require some new concepts and in my opinion this is also a bit out of the scope of Ki-nTree. This is actually a function which would perfectly fit the Inventree smartphone app.

lobra07, 2024-01-10:
there is an easy way to implement this .. every package from digikey have a dasta matrix by using any scanner into the search bar 
the search algorithm would detect that the input is data matrix and then make it own search using digikey part number from the data

eeintech, 2024-02-27:
For now I will close this as I don't think it makes sense to add too much complexity to Ki-nTree (and I don't have much bandwidth), we can re-evaluate later if anybody comes onboard willing to give it a shot.

closed 2024-02-27 by eeintech