sparkmicro/Ki-nTree #171: [support] initial categories setup

wilcomir, 2023-08-17:
Hello,
thanks for this nice piece of software.

I just started to use inventree, and currently have no categories set up. Checking out `categories.yaml`, I have a few questions as I am not sure I fully understand how this works.
My assumption is that Ki-nTree category options when creating a new part are read from the `yaml` file, rather than pulled from inventree itself.

1. Does the file support arbitrary nesting levels? I think that the answer is yes, but I want to make sure.
2. What happens if I create the categories using your script, and in some time I want to add/remove something? Can I just edit the file & rerun the script?
3. Can you clarify how the `CODES` section is used? It seems that you provide a three letter code for each "level 1" category - how is this used/what is expected from Ki-nTree about this? i.e. can I not define all `CODES`?

Thanks a lot for your time and patience.

eeintech, 2023-08-18:
Hello @wilcomir 

I'm glad you're enjoying Ki-nTree 😃 

`categories.yaml` can be setup by you with the categories structure you'd like to use in InvenTree and run the ` python -m kintree.setup_inventree` script to create them automatically. The same file can be automatically updated from InvenTree (assuming you have already setup your categories in InvenTree) using this option:

[image]

1. The file does support arbitrary nesting levels.
2. Yes you can do re-run the script, it won't create duplicate categories on same level. Maybe it is better for adding/removing categories from InvenTree directly then pull the updated structure using the button above.
3. `CODES` are solely used for IPN building (eg. the code will be part of your internal part number based on the top-level category code), you can disable their use here:

[image]

I hope this answer your questions, don't hesitate to ask!

wilcomir, 2023-08-23:
Hello @eeintech thanks for your clarifications. I was able to set up a category tree, and upload some parts.
On thing I have noticed, which might or might not be expected, is that even if I turn off the "Enable Category Codes" toggle, the tool still tries to parse the `CODES` section in `categories.yaml`. I had set it to `null`, but the parsing fails; in order to make it work I added a key: value pair under `CODES` and then everything works.

I am leaving this open to make sure you read this additional comment, if you feel that the behavior above is not expected I can open an issue about it.

eeintech, 2023-08-23:
This is not intentional behavior, I will need to fix that. Thanks for the feedback and let me know if you have any other question!

commit 98362056 (2023-10-14): Fix #171
PR #180 (2023-10-14): 1.0.4
* #172: Add supports for template handling for InvenTree version>=0.12.6 and fixes #165
* #173: Add price sanitization
* #175: Mouser parts use MPN for part name instead of description
* #179: Check that pricing data upload option is enabled by user
* Fix #171 bug with handling empty category `CODES`
closed 2023-10-14 by eeintech