sparkmicro/Ki-nTree #174: Cannot create part in categories having slash in the name

martonmiklos, 2023-09-19:
I have a category in inventree:
[image]
Uploading parts fails with:
[image]

eeintech, 2023-09-20:
@martonmiklos This is a very nice use case but thanks for reporting it. I would first suggest that you switch to a dash `-` instead of a slash, I believe `DC-DC` is actually the most common used name.

eeintech, 2023-10-10:
Unless you are attached to using slashes into categories (which is a bad idea as easily mixed-up with category tree), any chance the workaround proposed is good for you?

closed 2024-01-14 by eeintech