sparkmicro/Ki-nTree #178: Category does not exist error

HannesMorre, 2023-10-10:
I get this error when trying to use the category "Miscellaneous", but this category clearly does exist in inventree. I also reloaded inventree categories in Ki-nTree v1.0.3.

`[ERROR]    Category (['Miscellaneous']) does not exist in InvenTree`
[image]

HannesMorre, 2023-10-10:
All other categories are fine btw. I pulled a list of all categories with an API call on our inventree server and te "miscellaneous" category is present and anothing looks out of the ordinary.

Thieuzor, 2023-10-11:
I'm experiencing the same issue when creating a new part in the 'miscellaneous' category:
[image]

Same version information as above.
The category exists in Inventree:
[image]

eeintech, 2023-10-14:
Hello @HannesMorre @Thieuzor I am really confused because I don't have any issue with the `Miscellaneous` category using Ki-nTree v1.0.3 and InvenTree v0.12.7. I repeated it a couple of times with the `1490-NRF5340-DK-ND` part and got only good results:

``` shell
[MAIN]  Using Digi-Key cached data for 1490-NRF5340-DK-ND
[TREE]  Successfully connected to InvenTree server (ENV=DEVELOPMENT)
[INFO]  Warning: Parameter map for "Miscellaneous" does not exist or is empty
[INFO]  Success: Added new part to InvenTree
[INFO]  Warning: Image download failed (HTTP Error)

[MAIN]  Creating manufacturer part
[INFO]  Success: Added new manufacturer part

[MAIN]  Creating supplier part
[INFO]  Success: Added new supplier part

[MAIN]  Processing Price Breaks
[INFO]  Success: The price breaks were updated
```

I deleted the category and re-added it and it is still working fine... not sure how to reproduce this one 😬 

I might sounds stupid but have you checked for whitespace in the name on InvenTree GUI?

[image]

HannesMorre, 2023-10-16:
Good thinking, but no whitespace to be found in the name. Is it possible that we experience problems because Inventree is running in a docker?

[image]

eeintech, 2023-10-16:
I don't think Docker is the problem, because other categories are working for you. I find it extremely strange that you are two users experiencing the same issue but with the same setup I'm not able to... Have you tried Ki-nTree `1.0.4` which I released yesterday? There is a very low chance it fixes it, somehow, some way...

HannesMorre, 2023-10-17:
We tried with 1.0.4 but still the same issue. We added a subcategory to the miscellaneous category and when we create articles in this category, there are no issues.

eeintech, 2023-10-17:
> We added a subcategory to the miscellaneous category and when we create articles in this category, there are no issues.

That's good news, I also had a subcategory in Miscellaneous early on my testing and deleted it because I thought it could be the reason why it was working but it wasn't... I really have no clue how to help you further though 🤔

HannesMorre, 2023-10-18:
We'll work with subcategories for now. Thanks anyway for trying to help us 😄

closed 2024-02-03 by eeintech