sparkmicro/Ki-nTree #183: InvenTree HTTP Library support.

konkers, 2023-10-26:
KiCad 8.0 will have (already in `nightly`) HTTP library support (https://docs.kicad.org/master/en/eeschema/eeschema_advanced.html#http-libraries) which with the inventree_kicad plugin (https://github.com/afkiwers/inventree_kicad) allows one to tie InvenTree parts directly to KiCad symbols and footprints.  This has the advantage of not creating a descrete/separate KiCad symbol for each passive and instead link them all to their generic version (i.e. `Device:R_Small`).

It would be great to have a mode for Ki-nTree where it is aware of KiCad but instead of creating KiCad Symbols, instead just populates the `Symbol` and `Footprint` field in InvenTree.

konkers, 2023-10-27:
I built a rough prototype of this: https://github.com/konkers/Ki-nTree/tree/wip/magic

It drives the category, footprint, and symbol selection from a match file: https://github.com/konkers/Ki-nTree/blob/wip/magic/kintree/config/inventree/magic_settings.yaml

matmair, 2023-10-28:
Looks interesting @konkers; Not sure about the `magic` name for it - that leaves a lot of room for interpretation to the user.

Disclaimer: This is @eeintech project so my word has 0 sway here (nor should it, I am a newbie at KiCad at best)

konkers, 2023-10-28:
Oh yeah, the name magic is just a stand in.  The patch still needs work.   Notably the way the detection configuration file is structured, the user version is overwritten by the template data here: https://github.com/sparkmicro/Ki-nTree/blob/main/kintree/config/config_interface.py#L66.

konkers, 2023-10-28:
Another issue with the prototype I just found:  if one doesn't hit the magic button and instead manually enters the footprint and symbols, they are not added the the InvenTree part.  Probably something to do with `data_from_views` not getting updated.

eeintech, 2023-11-01:
Hello @konkers Sorry for the late response. This is super exciting development, I really need to look closer into it. Unfortunately my free time is quite limited right now. If you want to move forward and send a PR, when I get the chance I'll take a look into your implementation.

I also agree with @matmair the name `magic` is a bit marketing-ish, a more obvious name would be much better 😃

Nudelsalad, 2023-12-07:
Hi @konkers, 
I support your idea as Ki-nTree is a nice Desktop Tool to create parts. I already tested out your tool but I guess i have general setup problems with Ki-nTree so far. Did you  expand your magic Template or improve your fork?  Would love to see there a pr for the Http library mode inside Ki-nTree! Did you manage to solve the previous issue and how was your work with it so far?

eeintech, 2024-02-27:
Hello @konkers now that KiCad v8 is out, did you have time to polish your implementation?