sparkmicro/Ki-nTree #187: Error occurred during API request: Provided value is not a valid number

Weissnix4711, 2023-11-04:
I've been getting this issue and I'm not quite sure why. It implies the Temperature Range parameter has to be a valid number, but I have used Ki-nTree before to import many other parts from LCSC, with really weird values in that parameter like `+150℃@`. Not sure why the Tj keeps being dropped from the end of these strings, but whatever. Importantly, inventree took no issue with that in the past, but now it does.

I'm running inventree 0.12.0 and Ki-nTree 1.0.4

```
[MAIN]	LCSC search for C2828487

[MAIN]	Using LCSC cached data for C2828487
[TREE]	Successfully connected to InvenTree server (ENV=DEVELOPMENT)

[MAIN]	Using LCSC cached data for C2828487
[TREE]	Successfully connected to InvenTree server (ENV=DEVELOPMENT)
[INFO]	Warning: The following parameters were not found in supplier data:
['Vce Saturation (Max) @ Ib, Ic', 'Vgs(th) (Max) @ Id', 'Transistor Type', 'FET Type', 'Vgs (Max)', 'Supplier Device Package', 'Rds On (Max) @ Id, Vgs', 'Current - Collector (Ic) (Max)', 'Current - Continuous Drain (Id) @ 25°C', 'Power - Max', 'Power Dissipation (Max)', 'Voltage - Collector Emitter Breakdown (Max)', 'Drain to Source Voltage (Vdss)']
[INFO]	The following parameters are not mapped in LCSC parameters configuration:
['Continuous Drain Current (Id)', 'Drain Source On Resistance (RDS(on)@Vgs,Id)', 'Drain Source Voltage (Vdss)', 'Gate Threshold Voltage (Vgs(th)@Id)', 'Input Capacitance (Ciss@Vds)', 'Power Dissipation (Pd)', 'Reverse Transfer Capacitance (Crss@Vds)', 'Total Gate Charge (Qg@Vgs)', 'Type']

[MAIN]	Generating Internal Part Number
[INFO]	Internal Part Number = TRA-000153-00
[INFO]	Success: Added new part to InvenTree

[MAIN]	Creating parameters
Exception in thread Thread-737 (create_part):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/thomas/Projects/inventree/venv/lib/python3.11/site-packages/kintree/gui/views/main.py", line 1298, in create_part
    new_part, part_pk, part_info = inventree_interface.inventree_create(
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thomas/Projects/inventree/venv/lib/python3.11/site-packages/kintree/database/inventree_interface.py", line 612, in inventree_create
    if not inventree_process_parameters(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thomas/Projects/inventree/venv/lib/python3.11/site-packages/kintree/database/inventree_interface.py", line 697, in inventree_process_parameters
    parameter, is_new_parameter, was_updated = inventree_api.create_parameter(part_id=part_id, template_name=name, value=value)
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thomas/Projects/inventree/venv/lib/python3.11/site-packages/kintree/database/inventree_api.py", line 737, in create_parameter
    parameter = Parameter.create(inventree_api, {
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thomas/Projects/inventree/venv/lib/python3.11/site-packages/inventree/base.py", line 166, in create
    response = api.post(cls.URL, data, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thomas/Projects/inventree/venv/lib/python3.11/site-packages/inventree/api.py", line 420, in post
    response = self.request(
               ^^^^^^^^^^^^^
  File "/home/thomas/Projects/inventree/venv/lib/python3.11/site-packages/inventree/api.py", line 372, in request
    raise requests.exceptions.HTTPError(detail)
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url': 'https://redacted.net/api/part/parameter/', 'method': 'POST', 'status_code': 400, 'body': '{"data":["Provided value is not a valid number (°C)"]}', 'headers': {'AUTHORIZATION': 'Token 123redacted'}, 'params': {'format': 'json'}, 'data': {'part': 153, 'template': 50, 'data': '-55℃~+150℃@'}}
```

T0jan, 2023-11-06:
@Weissnix4711 Ki-nTree unfortunately does not work with InvenTree 0.12.0 because some limits in the parameter handling in this version. For this and various other reasons I would recommend to update to the latest InvenTree version (0.12.9 as I writing this) if you want to use Ki-nTree together with InvenTree. For more details on the bug see #165.

eeintech, 2023-11-14:
Hello @Weissnix4711 Did upgrading InvenTree work for you?

Weissnix4711, 2023-11-14:
Haven't had the chance to yet, sorry. Trying it now.

Weissnix4711, 2023-11-14:
Works fine now, thanks!

closed 2023-11-14 by Weissnix4711