sparkmicro/Ki-nTree #189: set up : conenxion xith inventree not realized

gerisse, 2023-11-08:
Hi
first , thank you for your job about this plugin , it is great !
i have a inventree droplet digitalocean, to test different plugins : the set up didn't connect to this server. Must i install kintree plugin also in the inventree settings server ?
Kintree , actually , is installed in my home pc : must be set up on the same computer , where inventree server is installed ?
So, you, see, it's not very clear for me ! if you can clear me ...
Thanks

eeintech, 2023-11-14:
Hello @gerisse sorry for the late response. You can install Ki-nTree on any computer, you'll have to enter the InvenTree server info into the settings menu: 

[image]

Then click on the "Test" button to check the connection to the InvenTree server can be made.

Let me know if that helps and if you have any other question!

gerisse, 2023-11-14:
Thanks but unfortunately no, like you can see :
[image]
i can give admin password  for this server, if you want, in PM
thanks

T0jan, 2023-11-15:
Can you test if you can talk with the inventree demo server?

URL: https://demo.inventree.org/
username: reader
Password: readonly

gerisse, 2023-11-16:
effectively, it s ok with demo server

eeintech, 2023-11-16:
> effectively, it s ok with demo server

In this case it may well be the connection to your DigitalOcean server, you could source help from them or the InvenTree community!

gerisse, 2023-11-17:
ok, thanks

closed 2023-11-17 by gerisse