sparkmicro/Ki-nTree #190: Error 500 from Inventree response when creating a part

markusEmm123, 2023-11-12:
After upgrading to Inventree 12.8 and Ki-nTree 1.0.4, Kintree halts with an error response from Inventree when i try to create a new part:

```
  File "C:\Users\username\AppData\Roaming\Python\Python310\site-packages\inventree\api.py", line 375, in request
    raise requests.exceptions.HTTPError(detail)
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url': 'https://inventree-server.tld/api/part/', 'method': 'POST', 'status_code': 500, 'body': '{"error":"TypeError","error_class":"<class \'TypeError\'>","detail":"Error details can be found in the admin panel","path":"/api/part/","status_code":500}', 'headers': {'AUTHORIZATION': 'Token 38438cfe8072f6e3c0edf19ce9df6b6d562e455e'}, 'params': {'format': 'json'}, 'data': {'name': 'MXC4005XC', 'description': 'Beschleunigungsmesser Monolithic and Wafer Level Packaged Three-Axis Accelerometer', 'category': 20, 'keywords': 'Beschleunigungsmesser Monolithic and Wafer Level Packaged Three-Axis Accelerometer', 'revision': 'A', 'active': True, 'virtual': False, 'component': True, 'purchaseable': True}}
```
(inventree-server.tld is the correct server name, i replaced it for privacy reasons; test connection to inventree server works).

Inventree shows an error

> Error object (253)
> Type:
> TypeError
> Path:
> /api/part/
> Info:
> expected string or bytes-like object
> When:
> 12. November 2023 20:08
> Data:
> Traceback (most recent call last):
> 
> File "/root/.local/lib/python3.9/site-packages/rest_framework/views.py", line 506, in dispatch
> response = handler(request, *args, **kwargs)
> 
> File "/root/.local/lib/python3.9/site-packages/rest_framework/generics.py", line 242, in post
> return self.create(request, *args, **kwargs)
> 
> File "/home/inventree/InvenTree/InvenTree/mixins.py", line 21, in create
> serializer.is_valid(raise_exception=True)
> 
> File "/root/.local/lib/python3.9/site-packages/rest_framework/serializers.py", line 227, in is_valid
> self._validated_data = self.run_validation(self.initial_data)
> 
> File "/home/inventree/InvenTree/InvenTree/serializers.py", line 253, in run_validation
> instance.full_clean()
> 
> File "/root/.local/lib/python3.9/site-packages/django/db/models/base.py", line 1236, in full_clean
> self.clean()
> 
> File "/home/inventree/InvenTree/part/models.py", line 835, in clean
> self.validate_ipn()
> 
> File "/home/inventree/InvenTree/part/models.py", line 602, in validate_ipn
> match = re.search(pattern, self.IPN)
> 
> File "/usr/local/lib/python3.9/re.py", line 201, in search
> return _compile(pattern, flags).search(string)
> 
> TypeError: expected string or bytes-like object

I'm not too sure where to look for further details an not even if this is caused by Inventree or Kintree.

I'm using Inventree 0.12.8 and Kintree 1.0.4

I see no IPN being transmitted to the Inventree API and that would somehow fit the error description in the Inventree error, but I'm not sure if that has to mean anything.

I would be very grateful for hints where to search for the cause or how to narrow down the problem so i can continue using this great piece of software :)

eeintech, 2023-11-14:
Hello @markusEmm123 Unfortunately I'm not able to reproduce, it works fine on my side, with the same software versions:

[image]

I thought it was because description/keywords field had too many chars but that's not the case.

What are your IPN settings?

[image]

[image]

markusEmm123, 2023-11-14:
IPN settings are identical to the ones in your screenshot.
Regex for IPN in Inventree is "([A-Z]{3,4})-([0-9]{6})-([0-9]{2})", so should match that.
Next free IPN is not yet in use.
I don't see any output in Ki-nTree; it seems like the program does not reach that stage in the program flow.

I also activated the "allow double IPNs" setting in Inventree, still the same error.
Removing parameters with "special" characters (e.g. the URLs to datasheets) did not change anything.
Replace IPN-Regex with .* (which should allow anything) still yields the same error.
Tick "Check for existing part" in Ki-nTree ends with the same error message.
Remove all "Checkbox" parameters in Inventree changes nothing in the behaviour.
Removing the units from the parameters in Inventree changes nothing.

I don't get a status display within Ki-nTree's main output window, only the always identical "HTTPError" in the background console.

Any ideas what else i could try?

T0jan, 2023-11-15:
I also can not reproduce this. I am running InvenTree 0.12.9 but this should not make a difference on the requests send out by Ki-nTree. Does the error only happens on this part or on others as well?

To debug this further we probably need to adjust the base code a bit to get more comprehensive debugging messages out. Do you have any experience with Python programming?

markusEmm123, 2023-11-16:
The error stays the same with other parts (from other Inventree categories, so with different part parameters).

I have some experience in Python programming.
The last request (payload) that is sent (through inventree\api.py) is
```
 - params: {'format': 'json'}
 - timeout: 10
 - headers: {'AUTHORIZATION': 'Token MyToken'}
 - auth: None
 - proxies: None
 - json: {'name': 'EP5WS10RJ', 'description': 'Drahtwiderstände - Durchgangsloch EP 5W (S) 10R 5%', 'category': 38, 'keywords': 'Drahtwiderstände - Durchgangsloch EP 5W (S) 10R 5%', 'revision': 'A', 'active': True, 'virtual': False, 'component': True, 'purchaseable': True}
```

That matches with the code 500 response code content.
I see no IPN in there but I assume that is defined after the part is initially created (so we know the part number).

I'm starting to think that i might have misconfigured something within Inventree and set a parameter to required which is not transmitted or something like that.

What debug output do we need?

T0jan, 2023-11-17:
> I'm starting to think that i might have misconfigured something within Inventree and set a parameter to required which is not transmitted or something like that.

I think this is it. I was able to reproduce your error after setting the `IPN Regex` under Inventree Settings --> Global Settings --> Parts (requires admin rights). If set this requires an IPN while creating a part while Ki-nTree only adds the IPN later as you observed correctly.

There could be ways around this, e.g. by generating the IPN before creating a part with the ID of the last part in the database but I think this could be quite low performant. What do you think @eeintech?

markusEmm123, 2023-11-17:
Hi, removing the Regex fixed the problem indeed.
But with the regex it worked up to 1.0.0rc with an older Inventree version (i think) 11.x or an early 12.x
Was there a change in the way Ki-nTree (resp. the Python interface) creates the parts or was the change on Inventree-server side?

T0jan, 2023-11-17:
this part of Ki-nTree was stable for at least this whole year. Considering InvenTree changed a lot on how the API interacts with the database this year, especially on the switch from 0.11.x to 0.12.0, I wouldn't be surprised if this is one of the consequences we hadn't seen yet.

markusEmm123, 2023-11-17:
Do we send an empty IPN or no IPN at all? Extending the Regex to accept an empty IPN should be possible. 
But the error message looks like we are not sending anything at all, if i had to guess...

Getting the highest current number and then creating a new article with article number +1 yields a risk for race condition at parallel usage. Not a problem for me at the moment, but might become one in the future.
Maybe we can send an empty IPN or dummy IPN and change the IPN afterwards.

T0jan, 2023-11-17:
Atm no IPN is send at all. If an empty IPN works while a IPN regex is set this could be an idea but I have not tested this. The only minor problem I could see is an error in the case there is already a part without an IPN in the database and the `Allow Duplicate IPN` setting in InvenTree is deactivated.

Dummy IPN has the problem that Ki-nTree needs to know the pattern of the regex, so it would probably need another user setting as the setting can be retrieved via API but only if the user has staff status and I don't want to limit Ki-nTree only to users with staff access.

markusEmm123, 2023-11-17:
> Dummy IPN has the problem that Ki-nTree needs to know the pattern of the regex, so it would probably need another user setting as the setting can be retrieved via API but only if the user has staff status and I don't want to limit Ki-nTree only to users with staff access.

I personally would find it acceptable to have (for a few seconds) a part which follows the IPN-scheme set up in Ki-nTree just with something like 000 or 999999 as part number.
If we want to adhere to the regex we would need to know it anyway (also with the current workflow). Right now it works to just let the user set up the Inventree and Ki-nTree configuration accordingly (except for the case where we try to send no IPN).

T0jan, 2023-11-17:
True, allowing the InvenTree regex to not match the IPN how it is currently generated by ki-nTree probably would need a complete rework of this aspect anyway and would be complicated to say at least. So we only support regex matching the structure of Ki-nTrees IPNs and add a note to the docs which makes this clear.

T0jan, 2023-11-20:
added the server regex support to my pull request. should be part of the next release.