sparkmicro/Ki-nTree #191: Error handling when adding parameters

markusEmm123, 2023-11-20:
For the parameter "package type" i have a limited list of values.
If the parameter from supplier does not match one of those values, the add-part-run stops without any useful error message.

How to reproduce:
Create a parameter template "package type" with the coma separated values "SMD, THT, Offboard".
Map that parameter template to a part category.
Create Mouser part 579-AVR64DD14-E/SL (it has "package type" set to "-") in said category.

Ki-nTree fails with the following error message and a half-created part (created in Inventree with IPN, name, picture, but no supplier and not at all in KiCAD).

```
This is the payload:
 - params: {'format': 'json'}
 - timeout: 10
 - headers: {'AUTHORIZATION': 'Token mytoken'}
 - auth: None
 - proxies: None
 - json: {'part': 294, 'template': 16, 'data': '-'}
This was the payload
Exception in thread Thread-158 (create_part):
Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\Program Files\Python310\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\username\AppData\Roaming\Python\Python310\site-packages\kintree\gui\views\main.py", line 1298, in create_part
    new_part, part_pk, part_info = inventree_interface.inventree_create(
  File "C:\Users\username\AppData\Roaming\Python\Python310\site-packages\kintree\database\inventree_interface.py", line 622, in inventree_create
    if not inventree_process_parameters(
  File "C:\Users\username\AppData\Roaming\Python\Python310\site-packages\kintree\database\inventree_interface.py", line 707, in inventree_process_parameters
    parameter, is_new_parameter, was_updated = inventree_api.create_parameter(part_id=part_id, template_name=name, value=value)
  File "C:\Users\username\AppData\Roaming\Python\Python310\site-packages\kintree\database\inventree_api.py", line 738, in create_parameter
    parameter = Parameter.create(inventree_api, {
  File "C:\Users\username\AppData\Roaming\Python\Python310\site-packages\inventree\base.py", line 166, in create
    response = api.post(cls.URL, data, **kwargs)
  File "C:\Users\username\AppData\Roaming\Python\Python310\site-packages\inventree\api.py", line 423, in post
    response = self.request(
  File "C:\Users\username\AppData\Roaming\Python\Python310\site-packages\inventree\api.py", line 375, in request
    raise requests.exceptions.HTTPError(detail)
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url': 'https://inventree.myserver.com/api/part/parameter/', 'method': 'POST', 'status_code': 400, 'body': '{"data":["Invalid choice for parameter value"]}', 'headers': {'AUTHORIZATION': 'Token mytoken'}, 'params': {'format': 'json'}, 'data': {'part': 294, 'template': 16, 'data': '-'}}

```

I would prefer a meaningful warning and continue the creation without that parameter.
Proposal: Put try-except around parameter-creation at the right place. I'm not sure where that is, though.
Maybe inventree_interface.py / somewhere around line 708?

T0jan, 2023-11-22:
I want to add another feature (mapping of standard images) in the next days. I will add the error handling for this case in between.

T0jan, 2023-11-27:
@markusEmm123 should be fixed in my pull request

closed 2024-01-16 by eeintech