sparkmicro/Ki-nTree #192: paramteres.yaml overwritten during startup

Fivefold, 2023-11-28:
I'm trying a fresh inventree + Ki-nTree setup and have the following problem:

When running `python3 -m kintree.setup_inventree` I get the following "Invalid physical unit" error during parameter creation:

```
Traceback (most recent call last):
  File "C:\Users\******\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\******\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\******\AppData\Local\Programs\Python\Python310\lib\site-packages\kintree\setup_inventree.py", line 54, in <module>
    setup_inventree()
  File "C:\Users\******\AppData\Local\Programs\Python\Python310\lib\site-packages\kintree\setup_inventree.py", line 46, in setup_inventree
    pk = inventree_api.create_parameter_template(name, unit)
  File "C:\Users\******\AppData\Local\Programs\Python\Python310\lib\site-packages\kintree\database\inventree_api.py", line 688, in create_parameter_template
    parameter_template = ParameterTemplate.create(inventree_api, {
  File "C:\Users\******\AppData\Local\Programs\Python\Python310\lib\site-packages\inventree\base.py", line 166, in create
    response = api.post(cls.URL, data, **kwargs)
  File "C:\Users\******\AppData\Local\Programs\Python\Python310\lib\site-packages\inventree\api.py", line 420, in post
    response = self.request(
  File "C:\Users\******\AppData\Local\Programs\Python\Python310\lib\site-packages\inventree\api.py", line 372, in request
    raise requests.exceptions.HTTPError(detail)
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url': 'https://********/api/part/parameter/template/', 'method': 'POST', 'status_code': 400, 'body': '{"units":["Invalid physical unit"]}', 'headers': {'AUTHORIZATION': 'Token *********************************'}, 'params': {'format': 'json'}, 'data': {'name': 'DC Resistance', 'units': 'mÎ©'}}
```

Note how it says `'units': 'mÎ©'` in the error message.

The offending line is https://github.com/sparkmicro/Ki-nTree/blob/7f894dd5f1d89bce128e022bcd39fa7fb603d890/kintree/config/inventree/parameters.yaml#L10 which gets changed to 
```yaml
DC Resistance: "m\xCE\xA9"
```
immediately upon running `python3 -m kintree.setup_inventree` (or simply running `kintree`) and before the parameter creation.

Also the two comment lines at the start get removed: https://github.com/sparkmicro/Ki-nTree/blob/7f894dd5f1d89bce128e022bcd39fa7fb603d890/kintree/config/inventree/parameters.yaml#L1-L2

**It seems the file actually gets overwritten**, because if I delete the offending lines and start kintree, I get them back (in the described state as above).

## Things I've tried

I had an old version of kintree from over a year ago installed so I thought it might be related to old config files. But removing the config folder and reinstalling kintree via pip to start from fresh did not help.

I also tried manually overwriting the `parameters.yaml` directly after startup without success. Seems it's already loaded into memory at that point.

## Version information:
Windows 10 Pro 22H2
Inventree 0.12.10
Ki-nTree 1.0.4

eeintech, 2023-11-29:
Hello @Fivefold 

Yes this is an intended behavior. I'm not sure about special characters like the Ω symbol, I have never tested it on InvenTree side. Does it work there?

eeintech, 2023-12-01:
I'm not sure if the YAML file can store the `Ω` symbol? 🤔 
It should be fine for you to set parameters in the `parameters.yaml` file, they shouldn't get overwritten. If they are then there is definitely a bug.

Fivefold, 2023-12-01:
That's the behaviour I'm experiencing. `parameters.yaml` always gets changed to the same faulty state upon start of Ki-nTree, as described above. I can make a video if needed.

I worked around this by creating the parameters by hand in InvenTree, so I personally don't urgently need a fix.

lobra07, 2024-01-10:
i have the same problem  to make it work i deleted the special chars and manually added them to inventree using the admin page

commit 1a385edc (2024-02-03): Fix #192
PR #208 (2024-02-03): 1.0.6
- #204 
- Fix #205 
- Fix #195 
- Fix #206
- Fix #192
closed 2024-02-03 by eeintech