sparkmicro/Ki-nTree #193: [HOW?] Could not convert to °C or mm

Nudelsalad, 2023-12-06:
I tried setting up kintree with a fresh install of inventree and used the default setting. I don't know if it is related to  #97 but my part fetched from digikey gets halfway created with this error:

```
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url':
https://inventree.example.de/api/part/parameter/', 'method': 'POST', 'status_code': 400, 'body': '{"data":["Could not convert - to
°C"]}', 'headers': {'AUTHORIZATION': 'Token inv-**********************-20231205'}, 'params': {'format': 'json'}, 'data': {'part': 7,
'template': 50, 'data': '-'}}
```
Same issue with the parameter field size if it receives the size from digikey e.g.: 1.2x3.4mm ->(could not convert to mm)
What am I doing wrong here since I am new to Kintree I guess its a setup failure. 
Your Readme explains that it is possible to use the fallback default settings for categories.yaml. I initialized my inventree instance with all your categories but somehow I got stuck here. 
Appreciate any help, thanks

T0jan, 2023-12-07:
Do you have the latest InvenTree version (0.12.10) running on your server? In some of the earlier version there were checks implemented which explicitly failed if parameter values were not numbers, see #165 for more details on that.

Nudelsalad, 2023-12-10:
I am currently running 0.13.0 the latest version.

closed 2023-12-10 by Nudelsalad