sparkmicro/Ki-nTree #194: Problem to load categories from Inventree

tnemrap, 2023-12-06:
When i try to reload categories from Inventree i get this error:

`[TREE]  Successfully connected to InvenTree server (ENV=DEVELOPMENT)
Error at api.request - GET @ http://astrid:8000/api/part/category/
Exception in thread Thread-51 (reload_categories):
Traceback (most recent call last):
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\connectionpool.py", line 790, in urlopen
    response = self._make_request(
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\connectionpool.py", line 536, in _make_request
    response = conn.getresponse()
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\connection.py", line 461, in getresponse
    httplib_response = super().getresponse()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\http\client.py", line 1375, in getresponse
    response.begin()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\http\client.py", line 318, in begin
    version, status, reason = self._read_status()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\http\client.py", line 279, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\socket.py", line 705, in readinto
    return self._sock.recv_into(b)
ConnectionResetError: [WinError 10054] En befintlig anslutning tvingades att st▒nga av fj▒rrv▒rddatorn

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\requests\adapters.py", line 486, in send
    resp = conn.urlopen(
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\connectionpool.py", line 844, in urlopen
    retries = retries.increment(
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\util\retry.py", line 470, in increment
    raise reraise(type(error), error, _stacktrace)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\util\util.py", line 38, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\connectionpool.py", line 790, in urlopen
    response = self._make_request(
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\connectionpool.py", line 536, in _make_request
    response = conn.getresponse()
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\urllib3\connection.py", line 461, in getresponse
    httplib_response = super().getresponse()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\http\client.py", line 1375, in getresponse
    response.begin()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\http\client.py", line 318, in begin
    version, status, reason = self._read_status()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\http\client.py", line 279, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\socket.py", line 705, in readinto
    return self._sock.recv_into(b)
urllib3.exceptions.ProtocolError: ('Connection aborted.', ConnectionResetError(10054, 'En befintlig anslutning tvingades att st▒nga av fj▒rrv▒rddatorn', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python310\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\kintree\gui\views\main.py", line 695, in reload_categories
    self.fields['Category'].options = self.get_category_options(reload=True)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\kintree\gui\views\main.py", line 684, in get_category_options
    for category in inventree_interface.build_category_tree(reload=reload)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\kintree\database\inventree_interface.py", line 86, in build_category_tree
    categories = inventree_api.get_categories()
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\kintree\database\inventree_api.py", line 105, in get_categories
    children = category.getChildCategories()
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\inventree\part.py", line 45, in getChildCategories
    return PartCategory.list(self._api, parent=self.pk, **kwargs)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\inventree\base.py", line 203, in list
    response = api.get(url=url, params=kwargs)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\inventree\api.py", line 524, in get
    response = self.request(url, method='get', **kwargs)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\inventree\api.py", line 340, in request
    raise e
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\inventree\api.py", line 332, in request
    response = methods[method](api_url, **payload)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\requests\api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\requests\api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\requests\sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\requests\sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\micha\OneDrive\Dokument\KinTree\env-kintree\lib\site-packages\requests\adapters.py", line 501, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'En befintlig anslutning tvingades att st▒nga av fj▒rrv▒rddatorn', None, 10054, None))
`

tnemrap, 2023-12-06:
Found it.
It is a problem when running InvenTree in production mode using gunicorn.
But when running Development server i have no issues.

T0jan, 2023-12-07:
Still not good tho, Ki-nTree should work with production servers as well. Do you have your development-server also running behind gunicorn?

eeintech, 2024-02-27:
Closing as OP said it was a setup issue.

closed 2024-02-27 by eeintech