sparkmicro/Ki-nTree #196: Get supplier parameters without using inventree

guatek, 2024-01-08:
I'm using Ki-nTree without inventree to create KiCAD symbols and it works great! However I'd like to populate more of the supplier fields into my KiCAD symbol (eg. Capacitance and Tolerance). Is there a way to access these parameters in the symbol template without first going through inventree? I haven't dug into the code yet, but any tips would be appreciated. Thanks!

T0jan, 2024-01-08:
hi @guatek,

it is indeed possible. You only have to setup the parameter mapping for each of your libraries individual in the `supplier_parameters.yaml` in your config files. An example for the library `test` would look like this:

```
test:
  ESR:
  - ESR (Equivalent Series Resistance)
  Package Height:
  - Height - Seated (Max)
  - Thickness (Max)
  Package Size:
  - Size / Dimension
  Package Type:
  - Package / Case
  Rated Voltage:
  - Voltage - Rated
  - Voltage Rated
  Temperature Grade:
  - Temperature Coefficient
  Temperature Range:
  - Operating Temperature
  Tolerance:
  - Tolerance
  Value:
  - Capacitance
```

With `ESR`, `Package Height` etc. being the parameter names in your template and ` - ESR (Equivalent Series Resistance)` etc. being the parameter names from Digi-Key.

guatek, 2024-01-08:
Thanks @T0jan! That works very well.

closed 2024-01-14 by eeintech