sparkmicro/Ki-nTree #198: Doesn't fetch prices from TME

tomse, 2024-01-14:
[image]

1.0.3 and 1.0.4 doesn't fetch prices from TME

T0jan, 2024-01-15:
to be expected as price break handling was not implemented in these versions yet. it will be in the upcoming 1.0.5 release.

tomse, 2024-01-28:
I've just tried v1.0.5, it seems it's still missing, or is there a bug?

[image]

T0jan, 2024-01-29:
It should work, but maybe something broke. let me check with the other bug together.

T0jan, 2024-01-29:
@tomse are you able to support me in debugging a bit? as an European I am not able to get a API-key from TME so I need someone to parse me the debug outputs etc.

T0jan, 2024-01-29:
@tomse fixed, for now in my fork: https://github.com/T0jan/Ki-nTree/tree/tme_bugs. I will merge it here and create a new version with the fix once I added a different feature I am working on rn.

tomse, 2024-01-29:
thanks a lot :-D

T0jan, 2024-02-05:
@tomse should be fixed in 1.0.6. please test if it works out for you as well and then close here.

tomse, 2024-02-05:
@T0jan thank you! both creating new part and updating existing works - awesome!!

also thank you for opening a tab with the component after adding/updating, I think that was added in 1.0.4 or 1.0.5.

closed 2024-02-05 by tomse

T0jan, 2024-02-05:
@tomse that was there for way longer already but you probably never noticed because something failed before Ki-nTree got there. but glad you like it :D