sparkmicro/Ki-nTree #199: Error in startup

G-Pereira, 2024-01-15:
Hey! I am trying to create a package definition for  kintree in Arch Linux
but although I am able to build it and install it I cannot run it
Error I get while running:
```
Traceback (most recent call last):
  File "/usr/bin/kintree", line 5, in <module>
    from kintree.kintree_gui import main
  File "/usr/lib/python3.11/site-packages/kintree/kintree_gui.py", line 1, in <module>
    import flet as ft
  File "/usr/lib/python3.11/site-packages/flet/__init__.py", line 1, in <module>
    from flet_runtime import *
  File "/usr/lib/python3.11/site-packages/flet_runtime/__init__.py", line 2, in <module>
    from flet_core.types import (
ImportError: cannot import name 'FLET_APP' from 'flet_core.types' (/home/gpereira/.local/lib/python3.11/site-packages/flet_core/types.py)
``` 
Package definition:
```
# Maintainer: Goncalo Pereira <goncalo_pereira@outlook.pt>
pkgname=python-kintree
_name=${pkgname#python-}
pkgver=1.0.4
pkgrel=1
pkgdesc="Fast part creation in KiCad and InvenTree"
url="https://github.com/sparkmicro/Ki-nTree"
depends=(
    'python'
    'python-digikey-api'
    'python-flet'
    'python-thefuzz'
    'python-inventree'
    'python-kiutils'
    'python-mouser'
    'python-multiprocess'
    'python-pyaml'
    'python-validators'
    'python-wrapt-timeout-decorator'
)
makedepends=(python-build python-installer python-wheel)
license=('GPL3')
arch=('any')
source=("https://files.pythonhosted.org/packages/source/${_name::1}/$_name/$_name-$pkgver.tar.gz")
sha256sums=('781f647df37d935cee6fba6535e166caaa9f496f0c668e9b61ceccda65310c20')

build() {
    cd "$srcdir/$_name-$pkgver"
    python -m build --wheel --no-isolation
}

package() {
    cd "$srcdir/$_name-$pkgver"
    python -m installer --destdir="$pkgdir" dist/*.whl
}
``` 
any idea what I might be doing wrong?

T0jan, 2024-01-15:
@G-Pereira you have the wrong version of Flet installed to run Ki-nTree. There were some changes between the latest pip version and the one used here which break Ki-nTree and for now nobody had time to fix them. If you use Flet for some other packages as well I would recommend you to setup a virtual environment for Ki-nTree and install it there, then the requirements should be correct.

G-Pereira, 2024-01-15:
Got it fixed. kintree is now available in Arch Linux!

closed 2024-01-15 by G-Pereira