sparkmicro/Ki-nTree #201: Digi-key package height incorrectly parsing

Con30, 2024-01-16:
When importing parts from Digi-key through the API, the Part Height is received in the following format: `0.110" (2.80mm)`
However, when this part is created in Inventree, it registers as Package Height `0.110\"`.

A Digi-key part in particular that this occurs with is 399-11404-1-ND
[image]
[image]
[image]

Terminal output:
```
[MAIN]  Digi-Key search for 399-11404-1-ND←[0m
←[92m[TREE]     Successfully connected to InvenTree server (ENV=DEVELOPMENT)←[0m
←[93m[INFO]     Warning: The following parameters were not found in supplier data:
['ESR (Equivalent Series Resistance)']←[0m
[INFO]  The following parameters are not mapped in Digi-Key parameters configuration:
['Packaging', 'Lead Style', 'Features', 'Applications', 'Lead Spacing', 'Ratings', 'Failure Rate', 'ECCN']
←[95m
[MAIN]  Generating Internal Part Number←[0m
[INFO]  Internal Part Number = CAP-000014-00
←[92m[INFO]     Success: Added new part to InvenTree←[0m
←[93m[INFO]     Warning: Image download failed (HTTP Error)←[0m
←[93m[INFO]     Warning: Image download failed (HTTP Error)←[0m
←[93m[INFO]     Warning: Image download returned the wrong file type←[0m
←[93m[TREE]     Warning: Failed to upload part image←[0m
←[95m
[MAIN]  Creating parameters←[0m
←[92m[INFO]     Success: The following parameters were created:←[0m
--->    Mounting Type
--->    Package Height
--->    Package Size
--->    Package Type
--->    Rated Voltage
--->    Temperature Grade
--->    Temperature Range
--->    Tolerance
--->    Value
--->    ESR
←[95m
[MAIN]  Creating manufacturer part←[0m
[TREE]  Creating new manufacturer "KEMET"
←[92m[INFO]     Success: Added new manufacturer part←[0m
←[95m
[MAIN]  Creating supplier part←[0m
←[92m[INFO]     Success: Added new supplier part←[0m
←[95m
[MAIN]  Processing Price Breaks←[0m
←[92m[INFO]     Success: The price breaks were updated←[0m
←[95m
```

eeintech, 2024-01-16:
Hello @Con30 Are you looking to keep both imperial and metric values? If not you could correct it before it gets created. Else I will need to check why InvenTree does not accept the space and/or add a connector char.

eeintech, 2024-01-18:
In this view you can edit it:

[image]

eeintech, 2024-02-27:
@Con30 Did it work out for you?

closed 2024-03-28 by eeintech