sparkmicro/Ki-nTree #203: Partnumber does not match & Part not found errors (using TME)

tomse, 2024-01-28:
I'm adding both -what I believe are- bugs here, as they seem to be very related.

I see from time to time Partnumber does not match, but I can still continue adding this to inventree
[image]

but I also have problems with part not found. MDC-206 and MDC-206E both report this error - this one cannot be added to inventree though
[image]

T0jan, 2024-01-29:
The second one is intended, as TME does not return a MPN for this part, I assume because they use different manufacturers for this generic pin header. For now Ki-nTree requires a manufacturer part number to be set, because it always generates/updates the connected manufacturer part entry in InvenTree. You could work around it by manually setting a manufacturer part number in the intended field.

I think we could support part creation without creating the manufacturer part if gets a regular problem. what do you think @eeintech?

T0jan, 2024-01-29:
The first one is also not a bug. The check compares the identifier you enter in the `Part Number` field with the `Manufacturer Part Number` field. For the given part these are not the same as the MPN contains additional spaces, so the check fails. So either remove the spaces to synchronize them or ignore the warning.

tomse, 2024-01-29:
thank you :-) 
a feature request - add some help text to what is wrong and maybe how it's fixed (maybe not on the same error message as some people aren't that fast readers before it disappears) - would/could be a fix in my eyes.. - leave it to the user how to handle it, or enter some kind of placeholder value to be ignored/edited on a later time?

T0jan, 2024-01-30:
I made the error messages a bit more descriptive in my fork, check if you still se issues with them now if you can.

After a bit of checking I am now opposed to adding support for parts without manufacturer number because too many other function (supplier part creation, pricing) directly depend on the manufacturer part in InvenTree and one would need to introduce a hacky workaround to get these still working.

tomse, 2024-01-30:
ok, thats understandable, thank you for looking into it.
love the application.. it makes it so much easier to add components :-D

closed 2024-01-31 by T0jan

eeintech, 2024-01-31:
Sorry I'm late to the party. Mandatory MPN is indeed something I wouldn't touch, especially that there is no use-case for it inside Ki-nTree, one can do it real quick directly inside InvenTree.
It could be that the warning/error messages are not explicit enough, apologize for that. I see that @T0jan has already proposed a fix!