sparkmicro/Ki-nTree #205: Incorrect supplier

Con30, 2024-01-30:
Digi-Key parts are failing to be created. The following message is seen when trying to create a new part: 
```
[MAIN]  Creating supplier part←[0m 
[TREE]  Creating new supplier "DIGI" 
Exception in thread Thread-149 (create_part): 
Traceback (most recent call last):
```
The Supplier Name is unchanged from what is being received from the Digi-Key API and is "Digi-Key" not "DIGI". Additionally, when the Supplier name is changed to something else the same message is seen saying it is trying tp create the supplier "DIGI". Clearing the cache and reinstalling kintree does not appear to resolve the issue.

Are there any other places it could be pulling this from?

T0jan, 2024-01-31:
Does this happens for specific parts or for all of them? Do you already have a supplier `Digi-Key` in your inventree database?

Con30, 2024-01-31:
This appears to be for all parts. Yes, the supplier `Digi-Key` already exists in the database.

I'm wondering if it is searching the companies table and finding the Manufacturer `DIGI` and trying to use it as it is close to `Digi-Key`.

eeintech, 2024-02-03:
Yup I was able to reproduce with a "DIGI" manufacturer, good catch and sorry for the issue, fixing it now...

commit 80c60b8c (2024-02-03): Fix #205
PR #208 (2024-02-03): 1.0.6
- #204 
- Fix #205 
- Fix #195 
- Fix #206
- Fix #192
closed 2024-02-03 by eeintech