sparkmicro/Ki-nTree #206: Getting the default values added to the configuration files

nikolai2111, 2024-02-01:
Hello everyone,

First of all, thank you very much for the great tool! It makes a lot of work easier.

Now I have the following problem, when I edit the following files and adapt them to my categories, the default values are still loaded in.
- digikey_categories.yaml
- parameters_filters.yaml
- supplier_parameters.yaml

How can I ensure that these default configurations are no longer loaded? The files could be found in the [categories.zip](https://github.com/sparkmicro/Ki-nTree/files/14124018/categories.zip).

In addition, would it be possible to give me feedback on whether the files have been created correctly or have been correctly adapted to my categories?

Thank you for your help

eeintech, 2024-02-03:
Hello @nikolai2111 

I'm glad you enjoy Ki-nTree, thank you for your kind words 😉 

Very sorry for this issue, it was actually related to #195 and I fixed it in #206.
It will be fixed with `1.0.6` 😃

closed 2024-02-03 by eeintech

nikolai2111, 2024-02-04:
Great, thank you very much for the release!

Do I understand the advanced settings correctly. That if these .yaml files are configured correctly, when I have entered an article from Digikey, that the settings in Ki-nTree under the tab InvenTree are set directly by Ki-nTree. Coresponding to my mapping in the file?

So that I would only have to look at this tab in part to check the consistency?

eeintech, 2024-02-06:
Those files are use for mapping yes, for files that are updated by the GUI (eg. https://github.com/sparkmicro/Ki-nTree/blob/c2f27feafe349c66eafccdddb83394aa9eb41859/kintree/config/config_interface.py#L31) you do not have to check for consistency, unless you're having an issue then let me know.
For files which are not managed by the GUI, I can give you more details case by case if you'd like, or if you have time you can checkout the code 😄

PR #208 (2024-02-03): 1.0.6
- #204 
- Fix #205 
- Fix #195 
- Fix #206
- Fix #192