sparkmicro/Ki-nTree #207: Problem creating a supplier from Digikey

nikolai2111, 2024-02-01:
Hello everyone,

First of all, thank you very much for the great tool! It makes a lot of work easier.

I have a problem with the integration of the TPS613222ADBVR component from Digikey. I have already been able to integrate two components, 1x a resistor and 1x a crystal.

The following errors are displayed in the PowerShell:
```
[MAIN]  Using Digi-Key cached data for 296-50503-1-nd←[0m
←[92m[TREE]     Successfully connected to InvenTree server (ENV=DEVELOPMENT)←[0m
←[93m[INFO]     Warning: The following parameters were not found in supplier data:
['Current - Quiescent (Iq)']←[0m
[INFO]  The following parameters are not mapped in DigiKey Corporation parameters configuration:
['ECCN', 'Function', 'Mounting Type', 'Number of Outputs', 'Output Configuration', 'Package / Case', 'Packaging', 'Synchronous Rectifier']
←[95m
[MAIN]  Generating Internal Part Number←[0m
[INFO]  Internal Part Number = PWR-000043-00
←[92m[INFO]     Success: Added new part to InvenTree←[0m
←[93m[INFO]     Warning: Image download failed (HTTP Error)←[0m
←[93m[INFO]     Warning: Image download failed (HTTP Error)←[0m
Exception in thread Thread-341 (create_part):
Traceback (most recent call last):
  File "C:\Users\nikol\AppData\Local\Programs\Python\Python311\Lib\threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "C:\Users\nikol\AppData\Local\Programs\Python\Python311\Lib\threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\nikol\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\gui\views\main.py", line 1299, in create_part
    new_part, part_pk, part_info = inventree_interface.inventree_create(
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nikol\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\database\inventree_interface.py", line 598, in inventree_create
    datasheet_link = inventree_api.upload_part_datasheet(inventree_part['datasheet'], part_pk)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nikol\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\database\inventree_api.py", line 393, in upload_part_datasheet
    if not download_with_retry(datasheet_url,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nikol\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\common\tools.py", line 124, in download_with_retry
    file = download(url, fileoutput=full_path, silent=silent, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nikol\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\common\tools.py", line 93, in download
    (file, headers) = urllib.request.urlretrieve(url, filename=fileoutput)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nikol\AppData\Local\Programs\Python\Python311\Lib\urllib\request.py", line 251, in urlretrieve
    tfp = open(filename, 'wb')
          ^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument: 'C:\\Users\\nikol\\kintree\\cache\\datasheets\\suppproductinfo.tsp?distId=10&gotoUrl=http%253A%252F%252Fwww.ti.com%252Flit%252Fgpn%252Ftps61322.pdf'
```

Ah and by the way how can the data sheets also be downloaded and saved in InvenTree? What needs to be configured for this?

eeintech, 2024-02-03:
Hello @nikolai2111 

I'm not able to reproduce your issue but I do see the local path to the datasheet may be posing a problem: `C:\\Users\\nikol\\kintree\\cache\\datasheets\\suppproductinfo.tsp?distId=10&gotoUrl=http%253A%252F%252Fwww.ti.com%252Flit%252Fgpn%252Ftps61322.pdf`

It's unfortunate but Texas Instruments parts on Digi-Key have a redirect/long URL which Ki-nTree can't download from... Please try replacing the URL before creating the part. For example for this part, replace the **Datasheet** field with: `https://www.ti.com/lit/ds/symlink/tps61322.pdf`

[image]

This works for me, please let me know!

nikolai2111, 2024-02-04:
Hmmm, that's special.

I tried your short link today and it works perfectly.
It is indeed the case that the TI links are very long. Could there be an algorithm that fixes the problem directly in software?

T0jan, 2024-02-05:
@eeintech do you think it would be a good idea to detect the TI links and replace them with the symlinks if they follow the outlined syntax?

T0jan, 2024-02-06:
@eeintech I wouldn't tie it to the supplier, as e.g. mouser also has this redirect thing to TI sometimes and I think other manufacturers do that as well. Instead I would go for some mechanism in the download-function to deal with these kinds of redirects, also for images etc. If we only want to do it for datasheets I would do it in the upload_datasheets function.

I also don't really see the reason to make it a user setting. There are so many other things going on in the background where the code would need an adjustment if the manufacturers/distributors change something on their websites (e.g. the headers for the html download request) and the user doesn't really care if it is a redirect or not.

> Or skip this entirely and append the lower case part number like @nikolai2111 mentioned, but would it work always?

This won't work for all parts either (e.g. SN74LVC2G17DBVR). I think the best bet is to detect the part number in the goToUrl and then construct the symlink URL with this. AT least with my limited library of TI parts this worked for 100% while the other ideas always had some error cases.

eeintech, 2024-02-09:
> I think the best bet is to detect the part number in the goToUrl and then construct the symlink URL with this. AT least with my limited library of TI parts this worked for 100% while the other ideas always had some error cases.

That was also my preferred option, so we can try to add this to the next minor release.

> @eeintech I wouldn't tie it to the supplier, as e.g. mouser also has this redirect thing to TI sometimes and I think other manufacturers do that as well. Instead I would go for some mechanism in the [...]

So you would use a generic lookup table that would find the "goToUrl" keyword and translate `http://www.ti.com/lit/gpn/ABC123.pdf` to `https://www.ti.com/lit/ds/symlink/ABC123.pdf`, is that the idea here?

T0jan, 2024-07-08:
fixed with v1.1.1

closed 2024-07-08 by T0jan