sparkmicro/Ki-nTree #209: charmap can't encode character

tomse, 2024-02-05:
Sorry, I'm not trying to discover bugs :-P 

adding/updating components from TME gives errors

`UnicodeEncodeError: 'charmap' codec can't encode character '\u03a9' in position 3: character maps to <undefined>`

Searching for "TSM112A1/TC" 

snippet from the end of the debug windows.. let me know if you need more

[image]

T0jan, 2024-02-06:
@tomse Am I right in assuming you are using kintree on some windows OS? Because I can only reproduce it on windows. In general the main development for Ki-nTree only happens on linux-based systems, so there are bugs to be expected on windows, as windows python packages have slight differences to the ones used in linux.

I will take a look at this one but there is certainly the option we won't fix some of the windows bugs if they interfere with the usability on linux.

T0jan, 2024-02-06:
@tomse So this already is such a case. There is something wrong in how one of the base modules handles the Ω in windows.

At least for this part you can work around it by disabling the `Supplier Search Cache` in the user settings. Sorry for the inconvenience, but windows...

eeintech, 2024-02-06:
@T0jan My setup is currently WSL (Ubuntu 22.04) so you're right about this, I can't reproduce this issue...

Maybe it has something to do with this: https://github.com/sparkmicro/Ki-nTree/blob/c2f27feafe349c66eafccdddb83394aa9eb41859/kintree/config/config_interface.py#L31

I added this option to solve #192

tomse, 2024-02-06:
@T0jan  correct.. using windows and python natively, I'm not running in WSL/Linux  

If you're trying to fix it by detecting OS, let me know, then I won't do the `allow_unicode` setting and wait till next release.

eeintech, 2024-02-09:
> @eeintech removing the `allow_unicode` solves the problem on windows. But I don't know if there is really a way to solve the unicode issue for both systems without adding some kind of os detection, because in my experience the unicode handling is always different between these two (and sometime even in between linux distros...)
> 
> For os detection I usually use something like this combined with a global flag of some sorts:
> 
> ```
> from sys import platform
> if platform == "linux" or platform == "linux2":
>     # linux
> elif platform == "darwin":
>     # OS X
> elif platform == "win32":
>     # Windows...
> ```

Ok good to know... I guess this is easy enough to implement for the next minor release, thanks!

PR #215 (2024-03-05): 1.0.7
* #212 
* #216 (fixes #151)
* Fix #209
closed 2024-03-05 by eeintech