sparkmicro/Ki-nTree #210: Python 3.12 support

xlxs4, 2024-02-06:
Have you looked into this? Besides `distutils` (removed in Python 3.12) used by `digikey-api` (https://github.com/peeter123/digikey-api/issues/48), is there any other dependency holding >3.11 support back?

eeintech, 2024-02-09:
I haven't looked into it yet, I haven't used Python 3.12 but I'm aware it has been out for a while now, I will check on the next release what's missing to get there, thanks!

AndronikosKostas, 2024-04-26:
any news on this?

eeintech, 2024-04-27:
@AndronikosKostas No news, the digikey API package hasn't been updated yet and I haven't had the chance to test Python 3.12 yet.

chl33, 2024-08-15:
This would be useful to me, as I just updated to Ubuntu 24.04 which uses Python 3.12.

eeintech, 2024-08-15:
It's next on my TODO list 😄 There is a PR (https://github.com/peeter123/digikey-api/pull/51) already in place, I need to test it out. It also involves the new Digi-Key API so it will take a bit more testing to verify everything is good. The `digikey-api` does not seem to be maintained anymore so maybe a fork would be needed soon...

T0jan, 2024-10-15:
As this gets now important for me as well as our IT wants to switch to 24.04 as well I would vote for replacing the digikey package with the fork. The maintainer hasn't done anything there in years so it's save to say he wont come back to enable 3.12 compatibility soon.

T0jan, 2024-10-15:
Also this will probably be a ubuntu 24.04 specific problem which needs to be solved: https://github.com/actions/runner-images/issues/10781

eeintech, 2024-10-15:
Hello @T0jan I may not have a lot of time on hands in the next few weeks, I did want to get it up and running on 3.12 at one point though... WSL 24.04 seems to be available for Windows now, that was also missing for my setup. I will start by checking that one out.

T0jan, 2024-10-15:
@eeintech if you need some help on the Linux side, I have some time to spare rn. Just tell me what you would need.

eeintech, 2024-10-15:
I was able to run Ki-nTree with both: 
* https://github.com/peeter123/digikey-api/pull/51
* https://github.com/peeter123/digikey-api/pull/47

So not with a single fork... and I haven't tried the Digi-Key API v4, did you?

T0jan, 2024-10-15:
@eeintech doesn't the first fork automatically use the v4 API? Or do you mean just checking if it works?

Also, I figured out how to get Flet and therefore Ki-nTree running on rolling release distros, if you want I can add a section on it to the README.

eeintech, 2024-10-15:
@T0jan The new Digi-Key API response is slightly different, so Ki-nTree would need to adapt to it. I had created an instance on Digi-Key's developer website to test it out but haven't had the chance to update the parsing yet.

> Also, I figured out how to get Flet and therefore Ki-nTree running on rolling release distros, if you want I can add a section on it to the README.

That'd be great!

T0jan, 2024-10-16:
@eeintech I will see what adaptations are needed for DigiKey v4 and fix the other minor bugs still open as well in the next days

eeintech, 2024-10-16:
@T0jan Awesome thanks 👍

eeintech, 2024-10-16:
Btw, there was a pending PR which I finally merged. Now that `1.1.4` is out, I think we should be aiming for `1.2.0` to enable Python 3.12 compatibility.

closed 2024-12-29 by eeintech

chl33, 2024-12-29:
Thanks for the update and fix! I got things working again on my Ubuntu 24.04 system.

To do this, I needed to do a couple tweaks after installing the wheel from the .zip file.

To fix an error about a missing shared library `libmvp.so.1`, I did:
```
sudo apt install libmpv-dev libmpv2
sudo ln -s /usr/lib/x86_64-linux-gnu/libmpv.so /usr/lib/libmpv.so.1
````
There is no `libmvp1` package available.  This is maybe questionable, but it got the app to start at any rate.

I then got an error about an missing user setting `OPN_USE_MANUFACTURER_PART_NUMBER` . So in my existing `~/.config/kintree/user/internal_part_number.yaml` I added
```
IPN_USE_MANUFACTURER_PART_NUMBER: true
```
and now the app runs.

eeintech, 2025-01-02:
Hello @chl33 Thanks for the report and steps to get `1.2.0` running.

I'm surprised you encountered an issue with a parameter, Ki-nTree tries to creates the missing on startup, maybe this one is not taken into account... I will open an issue to make sure it is in the future versions.

PR #270 (2024-12-29): 1.2.0
This is a port of #263 including a few tweaks and fixes.
- Fix #65 
- Fix #210 
- Fix #261
- Fix #264