sparkmicro/Ki-nTree #211: Flow to create kicad symbol and footprint?

peterus, 2024-02-16:
I am currently a little bit struggling with the flow to create a symbol and the correct footprint. What is the flow to do so?

- I am entering the part number, selecting the correct supplier and search for the part
- than I am navigating to the tab InvenTree and choose the correct category
- in the tab kicad I am selecting the library, symbol template and footprint library (all this are just empty libraries)
- now I am clicking on new footprint and I am entering a name for it
- on tab create i am just clicking on create part

I am getting now an symbol with some text inside (like the link to InvenTree, part name etc.) and no footprint.

Another try was to download the symbol and footprint from SnapEDA first and than select the correct footprint. The link will be made between an empty symbol and the footprint. But why not use the downloaded symbol?

I think I am still missing some important part in the flow to generate all this data.

T0jan, 2024-02-19:
hi @peterus,

The direct use of SnapEDA symbols is not supported rn. You either have to add it to your symbol templates (and update the fields accordingly) so that ki-nTree can use it or update the symbol after creation. My suggestion would be to not use SnapEDA symbols, as they are usually far from optimal. The best approach is to have high quality templates for all the standard components like Rs, Cs, etc. and create the symbols for ICs etc. after the data was filled into the empty template.

eeintech, 2024-02-27:
@peterus Please let us know if you're still stuck or it makes sense to you @T0jan's explanation 😃