sparkmicro/Ki-nTree #213: Ki-nTree flow to work together with KiCAD

NightSkySK, 2024-02-20:
This is not a bug report. I just can't figure out based on provided documentation what is designed workflow to work Ki-nTree with Kicad.

I've managed to set up API's and link Ki-nTree with Inventree. Parts are created as expected. 

Can I get some more explanation of what was expected to create those parts in KiCad?

1. In the KiCad settings (inside Ki-nTree) should I link to existing global libraries of symbols and footprints which are delivered with Kicad or there should be newly created library files to fill those libraries with new symbols and footprints ?
[image]

I assumed that it needs to be fresh/empty libraries where I will create new components, am I right?
What is the intended procedure after a new PN is created in the symbol library and footprint library?

I would appreciate if I could get a description of your workflow. I appreciate your work because even importing parts descriptions and parameters from suppliers is a huge support.

T0jan, 2024-02-21:
hi @NightSkySK,

in general I would not recommend to add your own parts to KiCads standard libraries, as they will get overwritten every time you do an update of KiCad and I think they are write protected on Windows anyway. If you don't want to start your libraries from scratch it could be an good idea to just copy the standard libraries somewhere else and set them as your global libraries in KiCad.

Apart from that there are not really any constrains on how you structure your own library when working with Ki-nTree. You can use only one single file to have all symbols or you also can have as many different library files based on whatever categorization you want.

It is also not necessary to have an empty library for adding new components, Ki-nTree can add components to existing libraries (of KiCad v7) without problems.

The general workflow is as follows:

* create an symbol template/adapt the ones Ki-nTrees ships with the install to your needs
* in Ki-nTree: select the symbol library you want to add the newly generated symbol to
* in Ki-nTree: select the symbol template you want to use as base for the newly generated symbol
* in Ki-nTree: select the footprint library and either an existing footprint or add the name of the new footprint
* in Ki-nTree: Create the part
* in KiCad, depending on your symbol template quality: adjust the newly generated symbol to the actual component (update drawings, pin numbers/identifiers etc.)
* in KiCad, if you created a new footprint and didn't downloaded the SnapEDA one: add pads, drawings etc. based on the component datasheet to the new, empty footprint.

A few words on the symbol templates: they are mostly there to add whatever data you want from the InvenTree part and its parameters also to the KiCad representation. You do this via the individual fields of the symbol and by adding the different identifiers like the parameter names in their values, Ki-nTree then will replace them during generation. It also can make sense to have the actual drawing of the symbol also in the template, for simple components like resistors, diodes, etc. For more complex parts like integrated circuits, opamps, MCUs etc. it makes no sense and you always have to do the pin assignments and drawings afterwards yourself. Ki-nTree at this point does not generate any complete symbol on its own.

NightSkySK, 2024-02-21:
Hi @T0jan Thank you for the explanation it's now clearer to me How to proceed with KiCad Libraries. I will play with it to find the best approach for myself.

eeintech, 2024-02-22:
Thanks @T0jan for the support!