sparkmicro/Ki-nTree #214: Categories map for other supplier like TME.

NightSkySK, 2024-02-21:
I would like to contribute to the project to help build categories and parametric mapping for other suppliers than Digikey.
I've started with TME categories mapping. Can someone help to confirm if I'm going in the right direction with this mapping:

here is the sample of capacitor mapping in `tme_categories.yaml`

```yaml
Capacitors:
  Aluminium:
  - Electrolytic capacitors
  - SMD electrolytic capacitors
  - THT electrolytic capacitors
  - SNAP-IN electrolytic capacitors
  Ceramic:
  - MLCC capacitors
  - MLCC SMD capacitors
  - MLCC THT capacitors
  - MLCC array capacitors
  - Ceramic capacitors
  - THT ceramic capacitors
  Polymer:
  - Polymer and Hybrid Capacitors
  - SMD Polymer Capacitors
  - THT Polymer Capacitors
  - SMD Hybrid Capacitors
  - THT Hybrid Capacitors
  - Polymer - tantalum capacitors
  Super Capacitors:
  - Supercapacitors
  Tantalum:
  - Tantalum capacitors
  - SMD tantalum capacitors
  - THT tantalum capacitors
```

so let's take as example https://www.tme.eu/pl/en/details/0805b104k500ct/mlcc-smd-capacitors/walsin/
it belong to the category:
[Passives](https://www.tme.eu/pl/en/katalog/passives_112309/)/[Capacitors](https://www.tme.eu/pl/en/katalog/capacitors_26/)/[MLCC capacitors](https://www.tme.eu/pl/en/katalog/mlcc-capacitors_100560/)/[MLCC SMD capacitors](https://www.tme.eu/pl/en/katalog/mlcc-smd-capacitors_113537/)

Is it correctly mapped in the above yaml file?

As I checked the source code `kintree/database/inventree_interface.py` method is `get_categories_from_supplier_data` is currently fixed to check only digikey categories mapping:
`category_map = config_interface.load_supplier_categories_inversed(supplier_config_path=settings.CONFIG_DIGIKEY_CATEGORIES)`

so to make it work with some other supplier categories mapping there need to be made some more changes here, isn't it?

eeintech, 2024-02-22:
The "auto-match" supplier category feature is quite old and I didn't give it much love as the GUI now let's you find InvenTree catgeories quite easily, at least in my opinion. If you really wish to get that magic for another supplier, it should be quite easy to add the `supplier` as an input parameter to the `get_categories_from_supplier_data` function and load the `category_map` accordingly. What you have in your file seems correct at first glance.

NightSkySK, 2024-03-02:
I will follow @T0jan recommendation to focus on `supplier_parameters.yaml` first, as that will give more benefits. After I look closer to the "auto-match" supplier category feature it becomes obvious that choosing a category from drop-down list is quite user-friendly as is.

closed 2024-03-02 by NightSkySK