sparkmicro/Ki-nTree #217: KeyError 'parameters' when updating to inventree

tomse, 2024-03-07:
kintree 1.0.7 (windows)

trying to update a component from lcsc creates this error with inventree 0.14.1

`[MAIN]  Searching for original part in database←[0m
←[92m[INFO] Success: Found original part in database (ID = 234 | Description = "TSSOP-20  ADC/DAC - Specialized ROHS")←[0m
←[93m[INFO]     Warning: The following parameters were not found in supplier data:
['Clock Frequency', 'Speed', 'Data Rate', 'Frequency Range', '-3db Bandwidth', 'Translator Type', 'Technology', 'Core Processor', 'Type', 'Sensor Type', 'Program Memory Size', 'Memory Size', 'Channels per Circuit', 'Package / Case', 'Voltage - VCCA', 'Voltage - VCCB', 'Voltage - Supply', 'Voltage - Supply (Vcc/Vdd)', 'Voltage - Supply, Digital', 'Voltage - Supply, Single (V+)', 'Operating Temperature']←[0m
Exception in thread Thread-174 (create_part):
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2288.0_x64__qbz5n2kfra8p0\Lib\threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2288.0_x64__qbz5n2kfra8p0\Lib\threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\kintree\gui\views\main.py", line 1380, in create_part
    alt_result = inventree_interface.inventree_create_alternate(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\kintree\database\inventree_interface.py", line 813, in inventree_create_alternate
    inventree_part = translate_form_to_inventree(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\kintree\database\inventree_interface.py", line 357, in translate_form_to_inventree
    for search_param in part_info['parameters'].keys():
                        ~~~~~~~~~^^^^^^^^^^^^^^
KeyError: 'parameters'
`

kintree 1.0.4 (ubuntu 20.04 - running in WSL on windows) (for some reason it wont upgrade using pip install kintree --upgrade)

`[MAIN]  Searching for original part in database
[INFO] Success: Found original part in database (ID = 234 | Description = "TSSOP-20  ADC/DAC - Specialized ROHS")
[INFO]  Warning: The following parameters were not found in supplier data:
['Clock Frequency', 'Speed', 'Data Rate', 'Frequency Range', '-3db Bandwidth', 'Translator Type', 'Technology', 'Core Processor', 'Type', 'Sensor Type', 'Program Memory Size', 'Memory Size', 'Channels per Circuit', 'Package / Case', 'Voltage - VCCA', 'Voltage - VCCB', 'Voltage - Supply', 'Voltage - Supply (Vcc/Vdd)', 'Voltage - Supply, Digital', 'Voltage - Supply, Single (V+)', 'Operating Temperature']
Exception in thread Thread-105:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/user/.local/lib/python3.8/site-packages/kintree/gui/views/main.py", line 1268, in create_part
    alt_result = inventree_interface.inventree_create_alternate(
  File "/home/user/.local/lib/python3.8/site-packages/kintree/database/inventree_interface.py", line 739, in inventree_create_alternate
    inventree_part = translate_form_to_inventree(
  File "/home/user/.local/lib/python3.8/site-packages/kintree/database/inventree_interface.py", line 313, in translate_form_to_inventree
    for search_param in part_info['parameters'].keys():
KeyError: 'parameters'
`

T0jan, 2024-03-11:
hi @tomse 

I can't reproduce it generally with v1.0.7. Please try first to install v1.0.7 and give it another shot afterwards, as there were various bugs between 1.0.4 and 1.0.7.

If it is still not working afterwards please provide more info, at least which part in LCSC causes the problems.

tomse, 2024-03-11:
> hi @tomse
> 
> I can't reproduce it generally with v1.0.7. Please try first to install v1.0.7 and give it another shot afterwards, as there were various bugs between 1.0.4 and 1.0.7.
> 
> If it is still not working afterwards please provide more info, at least which part in LCSC causes the problems.

The first error is running 1.0.7 on windows (so I am running the latest version), the second is just for reference..  the item was added using kintree 1.0.6 without problems.

this is updating the item and not adding as a new item

LCSC part: C131154

T0jan, 2024-03-11:
confirmed as bug with 0.14.1.

tomse, 2024-03-12:
Thanks again :-D

PR #221 (2024-04-30): 1.1.0
- #219 
- Fixes #217 
- Should fix #202 
- Update `flet` package to latest version #224 
- Incorporated #223 here directly as new `flet` version broke it
- Update README and add note in the docs about #225
closed 2024-04-30 by eeintech