sparkmicro/Ki-nTree #218: add part on Inventree , without kicad update

gerisse, 2024-03-09:
Hi
To be clearest , an example : 
i bought this part  (https://www.digikey.fr/fr/products/detail/texas-instruments/SN74LS393DR/1590637) : i want add it on my inventree instance , but not really on kicad , because it 's already present in schematic std lib and in footprint lib , with SOIC14.
is it possible to do this ? not add items in kicad, only in inventree ?
Thanks.
Best regards.

T0jan, 2024-03-11:
hi @gerisse,

yes, just make sure the kicad switch in the kicad tab is disabled before you create the part, then no kicad symbol will be generated.

[image]

gerisse, 2024-03-11:
oh, yes ! i didn't see it . Thanks!

closed 2024-03-11 by gerisse