sparkmicro/Ki-nTree #220: Setup Problems

StecklerCircuits, 2024-03-22:
I'm guessing/hoping this is a setup problem between Ki-nTree and Inventree.
The Invnetree install is in a docker container, and seems to be running ok.
It's been restarted a few times to make sure the data persists between docker boot ups.
Ki-nTree seems to be mostly working; I'm able to do searches from both digikey and mouser.
but I can't seem to get Ki-nTree to connect to Inventree.  what settings in either place should i be looking at?

[image]

StecklerCircuits, 2024-03-24:
any debug steps, or things to check here?

T0jan, 2024-03-25:
The culprit most likely is, that your system doesn't have the right name resolution for the localhost address. Try if it works if you change it to the localhost IP (`127.0.0.1`) plus port you made the server listen to.

For a more detailed debug output you can also open a python console and run following short code:

```
import requests
url = 'http://inventree.localhost'
response = requests.get(url, timeout=10)
```

StecklerCircuits, 2024-03-25:
well, that was enough to get me to start poking around.  
one thing that was helpful was the logfiles from Docker (for windows) the log files showed me that there might be a proxy running.  so i went and started looking for other IP addresses in different places. 
So at least it connects now!  using the proxy line. 
maybe someone else can explain all the "whys" here...  
now on to figuring out the category sync problem.

[image]

T0jan, 2024-03-26:
I never thought the proxy feature would be used for a localhost server but glad it helps :sweat_smile: 

As for why it is running through a proxy: no idea, that's a docker configuration thing.

> now on to figuring out the category sync problem.

if you can established the connection to the server it should work. Depending on how many categories you have it may take a few minutes. You also need to make sure the user you use for Ki-nTree can actually see the categories.

If you have any error messages or other problems feel free to post them here.

eeintech, 2024-03-28:
That's a weird one! I'm not a network expert myself but glad the proxy feature kicked-in to save the day. Do you mind if we close this issue?

closed 2024-04-02 by eeintech

rupraph, 2024-12-14:
Hi There,

I am sorry to post on a closed issue, but I have a setup issue to, and it might help others.

I am running Inventree from docker desktop on MacOS but can't get the Ki-nTree to connect to it. 
I tried using both http://invetree.localhost and 127.0.0.1 as well as adding the port number.
I am a bit confused on all the network virtualisation running behind the scenes when using docker.
when getting the polling docker info I got this:
[image]

Anybody with an hint ? 
Thanks !

T0jan, 2024-12-14:
Hi @rupraph 

Have you checked if docker maybe created a proxy here as in the other case as well?

How do you reach inventree via browser on the same machine?

rupraph, 2024-12-14:
Regarding the proxy question, what I understand from the case previous above is that the docker vm on windows did create a virtual network adapter, which was used as the "proxy" and worked for some reason. 

In my case I don't know what would be the address to put in the proxy field, the docker vm does not seem to add a virtual network adapter.

Regarding the address in the browser, I can reach it through http://inventree.localhost, but apparently Ki-nTree can't ?

eeintech, 2024-12-17:
Umm unsure how to help you here @rupraph as I'm no network expert... maybe ask in the InvenTree repo discussion space?

rupraph, 2025-03-16:
> Umm unsure how to help you here [@rupraph](https://github.com/rupraph) as I'm no network expert... Ki-nTree relies on the Python API provided by the InvenTree team, maybe ask in the InvenTree repo discussion space?

But how do people run Ki-ntree and inventree on mac if there is no default working solution when Inventree runs in a  container  ? Are most people running bare metal on linux ?

T0jan, 2025-03-17:
@rupraph I would suspect that most people don't run inventree und Ki-nTree on the same machine but in a (local) network which makes this problem doesn't appear. I tested it with docker in a single machine solution on linux a while back and it did work but there were a lot changes on both Ki-nTree and invetree in between so I can't guarantee it.