sparkmicro/Ki-nTree #222: Flet>=0.7.4,<0.8 requirement version does not exist

zisi, 2024-04-10:
Hey! I can not install Ki-n-tree in virtual env due to missing dependency in Flet package.
The specific versions Flet>=0.7.4,<0.8 do not exist in PyPi, https://pypi.org/project/flet/#history

Thanks

T0jan, 2024-04-10:
urgh, thats a major problem as flet 0.8 and newer break the GUI. So there is a major rework needed before this will be fixed to work with PyPi again. 

Not sure if an installation of older package versions directly from github releases is possible, this could be a solution in the meantime.

eeintech, 2024-04-10:
Bummer. I was planning on trying to update `flet` for release `1.1` but never got the chance to get there.

I still don't understand the decision (if it was) to remove those older versions of `flet`. I have opened an issue on their Github page: https://github.com/flet-dev/flet/issues/3014

eeintech, 2024-04-10:
@zisi One temporary fix but I think it will only work on Linux/Ubuntu:
- Download the wheel files attached below (ZIP file)
[flet_core-0.7.4-py3-none-any.zip](https://github.com/sparkmicro/Ki-nTree/files/14938064/flet_core-0.7.4-py3-none-any.zip)
- Unzip
- In your python environment, install both wheels running:
`pip install flet_core-0.7.4-py3-none-any.whl`
and then:
`pip install flet-0.7.4-py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl`
- Install kintree using pip: `pip install kintree==1.0.7`

Hopefully that works for you (and others) as a temporary measure.

zisi, 2024-04-11:
Thanks @eeintech !! It works!!

eeintech, 2024-04-12:
@zisi Great to hear! I'll keep this issue open as a reminder to update the `flet` package version...

second-string, 2024-04-20:
[kintree-1.0.99-py3-none-any.whl.zip](https://github.com/sparkmicro/Ki-nTree/files/15049488/kintree-1.0.99-py3-none-any.whl.zip)
For anyone on apple silicon, here's a wheel of Ki-nTree that I built using the open PR that supports the upgrade to flat 0.22.
```
python3 -m venv venv
source venv/bin/activate
pip install path/to/unzipped/wheel
```

dgossa, 2024-04-22:
I see workarounds for Linux and Mac. Any suggestions for Windows?
Thanks all.

dimoustroufis, 2024-04-22:
Hello, thanks for the fix ! I would like to mention that on Ubuntu 22.04 with python 3.11.9  I was able to install kintree 1.0.7 but when running it, the gui returned an error:  "IPN_ENABLE_CREATE". However, I was able to install and run kintree 1.0.5 without problems.

eeintech, 2024-04-22:
> I see workarounds for Linux and Mac. Any suggestions for Windows? Thanks all.

My proposed workaround works with Git Bash on Windows.

eeintech, 2024-04-30:
I'm happy to announce that `1.1.0` should solve this issue, please update! 😃

closed 2024-04-30 by eeintech