sparkmicro/Ki-nTree #225: InvenTree server field assumes https unless http specified

second-string, 2024-04-20:
No issue here, just adding info here in case someone else struggles with this. I was trying to connect my local Ki-nTree instance to a remote InvenTree server that's currently running without SSL certs by putting the remote URL in the Server Address field in the settings. The connection was failing with
```
[TREE]  Error connecting to InvenTree server: invalid address, username or password
```
 when it was just the url, but adding the `http://` prefix made everything play nicely.

[image]

second-string, 2024-04-20:
Closing cause no issue.

closed 2024-04-20 by second-string

eeintech, 2024-04-21:
Good point... should add this to the doc as it may not be obvious for everybody.