sparkmicro/Ki-nTree #227: Add additional parameters returned by API request

hillsandales, 2024-05-16:
Is there a way to import additional parameters returned by an API request?

Digikey returns more data than Mouser, but the descriptions for certain parts are better from Mouser. Mouser has the information needed in the attribute list on their website, and some of the data is returned through an API call (as seen using Postman), but the data is not showing up in Ki-nTree  when the toggle Parameters is switched to view Parameters.

I found a lot of config files, and have added the Parameter names I have setup in Inventree, and have tried to map the Parameter name to the name from Mouser, but it Ki-nTree doesn't seem to find the values when the API is called from within Ki-nTree. For instance, a resistor's RoHS status value is not found but it does show up in the Postman response as `"ROHSStatus": "RoHS Compliant"`.

Some of this data doesn't seem to be returned by Mouser's API though...

I get messages in the terminal like this:
```
[MAIN]  Using Mouser cached data for 560112116004
[TREE]  Successfully connected to InvenTree server (ENV=DEVELOPMENT)
[INFO]  Warning: The following parameters were not found in supplier data:
['Tolerance', 'Supplier Device Package', 'Power (Watts)', 'Operating Temperature', 'Resistance']
[INFO]  The following parameters are not mapped in Mouser parameters configuration:
['Packaging', 'Standard Pack Qty']
[INFO]  Success: Added new part to InvenTree
[INFO]  Warning: Image download socket timed out (3s)

[MAIN]  Creating parameters
[INFO]  Success: The following parameters were created:
--->    Temperature Range
--->    Value
--->    Footprint
[TREE]  Warning: The following parameters were skipped:
--->    Tolerance
--->    Package Type
--->    Rated Power
--->    Symbol
```

Being able to pull more parameters for types of parts would be really helpful!

T0jan, 2024-05-21:
@hillsandales sadly your observations are complete in this case. The mouser API does not return most of the parameters/info they list on the website, a full list of what they return can be found on their website: https://eu.mouser.com/api-search/

For the moment there is only standard processing of parameters available with Ki-nTree as the special formatting to extract info like the RoHS status is not yet implemented. If there is demand we could probably implement it, even better would be if someone could add a pull request for for it.

T0jan, 2024-10-21:
@hillsandales I checked once again and there is already a way to get additional parameters Mouser provides: In the mouser_config.yaml is a field `EXTRA_FIELDS` which can be used to treat any field in the Mouser data as parameter. Just add the `ROHSStatus` there and it will be added as a parameter if you have mapped it in the other config files accordingly.