sparkmicro/Ki-nTree #228: Kintree v1.1.0 does not startup

HannesMorre, 2024-05-21:
I just updated to kintree 1.1.0 and I only get an error on startup

```
(kintree-env) C:\Users\hmo\kintree>kintree
Unhandled error processing page session : Traceback (most recent call last):
  File "C:\Users\hmo\kintree\kintree-env\lib\site-packages\flet_runtime\app.py", line 242, in on_session_created
    await asyncio.get_running_loop().run_in_executor(
  File "C:\Users\hmo\AppData\Local\Programs\Python\Python39\lib\concurrent\futures\thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\hmo\kintree\kintree-env\lib\site-packages\kintree\gui\gui.py", line 56, in kintree_gui
    inventree_settings_view = InvenTreeSettingsView(page)
  File "C:\Users\hmo\kintree\kintree-env\lib\site-packages\kintree\gui\views\settings.py", line 859, in __init__
    super().__init__(page)
  File "C:\Users\hmo\kintree\kintree-env\lib\site-packages\kintree\gui\views\settings.py", line 323, in __init__
    self.fields[field_name].value = self.settings[field_data[0]]
KeyError: 'IPN_USE_MANUFACTURER_PART_NUMBER'
```

[image]

T0jan, 2024-05-21:
@HannesMorre I can reproduce it. It's most likely a consequence of a missing setting in one of your config files. Either replace the `user/internal_part_number.yaml`  with this one: https://github.com/sparkmicro/Ki-nTree/blob/main/kintree/config/user/internal_part_number.yaml

Or, if you don't want to overwrite your own settings, open your `user/internal_part_number.yaml` and add the line:

```
IPN_USE_MANUFACTURER_PART_NUMBER: false
```

HannesMorre, 2024-05-21:
@eeintech can you fix this in the next release?

HannesMorre, 2024-05-21:
@T0jan That works, thanks!

eeintech, 2024-05-21:
> @eeintech can you fix this in the next release?

Sure this issue could be prevented, reopening issue to not forget.

T0jan, 2024-05-22:
@eeintech should be doable with hard coding default states in the settings loading routine but maybe there is an more elegant, not so maintenance heavy way?

eeintech, 2024-05-22:
@T0jan The default parameters should be loaded from the `.yaml` files in the repo so maybe that setting was missing. Any new parameter should also be added to those files to prevent errors like this. I don't see any reason _not_ to hardcode it though.

T0jan, 2024-05-23:
@eeintech if it loads the repo config files than this shouldn't be at all and would be better to maintain, as it would be only one point of required changes. however, are you sure this is happening rn? last time I checked the settings code it was only loading the local config files.

eeintech, 2024-05-23:
@T0jan It should be done with this section of the code: https://github.com/sparkmicro/Ki-nTree/blob/a5af2f83926b6d424d6b8113bdccaf255a62de55/kintree/config/config_interface.py#L71

T0jan, 2024-05-24:
@eeintech Ah I see but yeah, like this it doesn't really help if there was a new setting introduced in the last update. Maybe this check should loop through the loaded parameters instead and merge/add the entries individually?

eeintech, 2024-05-24:
> @eeintech Ah I see but yeah, like this it doesn't really help if there was a new setting introduced in the last update. Maybe this check should loop through the loaded parameters instead and merge/add the entries individually?

Could be done yeah, probably more robust!

eeintech, 2024-08-07:
I think this was fixed with #251, closing.

closed 2024-08-07 by eeintech