sparkmicro/Ki-nTree #230: Datasheets from URL with a "?" in their name causes errors and freeze on Windows

sstteevvee, 2024-05-24:
Hi,

I'm using Python 3.9.9 on Windows 10, Ki-nTree version 1.1.0:
 - Settings > Inventree > "upload datasheets to inventree" enabled.
 - Downloading part info from Digikey
 - Creating new part in Inventree
 - Kicad features disabled

I've found that when the datasheet URL contains a question mark "?", that Ki-nTree generates a few errors messages and gets stuck. More-or-less of the component generation in Inventree succeeds.

Sample part: Digikey `296-35014-ND`
Problem URL: `https://www.ti.com/general/docs/suppproductinfo.tsp?distId=10&gotoUrl=https%3A%2F%2Fwww.ti.com%2Flit%2Fgpn%2Ftps54360`

Console log is below. The "create part" page's progress bar gets stuck half way, and the "cancel" button doesn't work. I need to close and restart the program to continue.

My workaround for now is to just clear out problematic datasheet URLs on the part search screen, before creating the part. My guess at the underlying problem is that Windows doesn't allow certain characters in filenames.

Please let me know if you'd like any more info.

- Steve.

```C:\Users\Steve>kintree
←[95m
[MAIN]  Digi-Key search for 296-35014-ND←[0m
←[92m[TREE]     Successfully connected to InvenTree server (ENV=DEVELOPMENT)←[0m
←[93m[INFO]     Warning: The following parameters were not found in supplier data:
['Current - Quiescent (Iq)']←[0m
[INFO]  The following parameters are not mapped in Digi-Key parameters configuration:
['Packaging', 'Mounting Type', 'Number of Outputs', 'Function', 'Output Configuration', 'Synchronous Rectifier', 'ECCN']
←[95m
[MAIN]  Generating Internal Part Number←[0m
[INFO]  Internal Part Number = PWR-0098
←[92m[INFO]     Success: Added new part to InvenTree←[0m
←[93m[INFO]     Warning: Image download failed (HTTP Error)←[0m
Future exception was never retrieved
future: <Future finished exception=OSError(22, 'Invalid argument')>
Traceback (most recent call last):
  File "C:\Users\Steve\AppData\Local\Programs\Python\Python39\lib\concurrent\futures\thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\Steve\AppData\Local\Programs\Python\Python39\lib\site-packages\flet_core\page.py", line 528, in wrapper
    handler(*args)
  File "C:\Users\Steve\AppData\Local\Programs\Python\Python39\lib\site-packages\kintree\gui\views\main.py", line 1442, in create_part
    new_part, part_pk, part_info = inventree_interface.inventree_create(
  File "C:\Users\Steve\AppData\Local\Programs\Python\Python39\lib\site-packages\kintree\database\inventree_interface.py", line 651, in inventree_create
    datasheet_link = inventree_api.upload_part_datasheet(inventree_part['datasheet'], part_pk)
  File "C:\Users\Steve\AppData\Local\Programs\Python\Python39\lib\site-packages\kintree\database\inventree_api.py", line 477, in upload_part_datasheet
    if not download_with_retry(datasheet_url,
  File "C:\Users\Steve\AppData\Local\Programs\Python\Python39\lib\site-packages\kintree\common\tools.py", line 124, in download_with_retry
    file = download(url, fileoutput=full_path, silent=silent, **kwargs)
  File "C:\Users\Steve\AppData\Local\Programs\Python\Python39\lib\site-packages\kintree\common\tools.py", line 93, in download
    (file, headers) = urllib.request.urlretrieve(url, filename=fileoutput)
  File "C:\Users\Steve\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 249, in urlretrieve
    tfp = open(filename, 'wb')
OSError: [Errno 22] Invalid argument: 'C:\\Users\\Steve\\kintree\\cache\\datasheets\\suppproductinfo.tsp?distId=10&gotoUrl=https%3A%2F%2Fwww.ti.com%2Flit%2Fgpn%2Ftps54360.pdf'
```

T0jan, 2024-05-24:
this is directly connected to this issue here: https://github.com/sparkmicro/Ki-nTree/issues/207

We could add a small fix which only changes the file name but as it anyway wouldn't download anything then I would rather fix the whole process.

sstteevvee, 2024-05-25:
A bit more testing.. After switching to a Linux machine (Ubuntu), I was able to successfully download some parts with a question mark in their datasheet URLs, e.g. `SSM3K376RLFCT-ND`. Still no luck with the TI parts, like in #207.

T0jan, 2024-07-08:
fixed in v1.1.1

closed 2024-07-08 by T0jan