sparkmicro/Ki-nTree #231: LCSC supplier not working

G-Pereira, 2024-05-24:
The API endpoint seems to be down. Is there any new API endpoint?

T0jan, 2024-05-24:
The old endpoint is definitely not working anymore, I was not able to find a new working one yet

30350n, 2024-05-24:
@zenermerps recently tracked this down and fixed it in [inventree_part_import#47](https://github.com/30350n/inventree_part_import/issues/47) and [inventree_part_import#48](https://github.com/30350n/inventree_part_import/pull/48).

It basically just changed from `https://wmsc.lcsc.com/wmsc/` to `https://wmsc.lcsc.com/ftps/wm/`.

T0jan, 2024-05-24:
@30350n perfect, will be fixed in the next release. Until then everyone can set the API-URL in the Ki-nTree settings manually to: 

`https://wmsc.lcsc.com/ftps/wm/product/detail?productCode=`

G-Pereira, 2024-05-24:
Thanks guys. I am so happy to continue using this tool with this fix.

closed 2024-05-24 by G-Pereira