sparkmicro/Ki-nTree #234: LCSC parts creation issues

NightSkySK, 2024-06-03:
I found that some parts from LCSC doesn't have parameters `paramVOList` which crashing Ki-nTree:

```bash
[MAIN]  LCSC search for C2997275
Future exception was never retrieved
future: <Future finished exception=KeyError('paramVOList')>
Traceback (most recent call last):
  File "C:\Python\Python311\Lib\concurrent\futures\thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\KSM\kintree\env\Lib\site-packages\flet_core\page.py", line 528, in wrapper
    handler(*args)
  File "C:\KSM\kintree\env\Lib\site-packages\kintree\gui\views\main.py", line 330, in run_search
    part_supplier_info = inventree_interface.supplier_search(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\KSM\kintree\env\Lib\site-packages\kintree\database\inventree_interface.py", line 480, in supplier_search
    part_info = lcsc_api.fetch_part_info(part_number)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\KSM\kintree\env\Lib\site-packages\kintree\search\lcsc_api.py", line 99, in fetch_part_info
    for parameter in range(len(part[parameter_key])):
                               ~~~~^^^^^^^^^^^^^^^
KeyError: 'paramVOList'
```

[API return](https://wmsc.lcsc.com/ftps/wm/product/detail?productCode=C2997275)
<details>
 <summary>JSON response </summary>

```JSON
{
  "code": 200,
  "msg": null,
  "result": {
    "currencyType": "USD",
    "isAutoOffsale": false,
    "productArrange": "Bag-packed",
    "productModel": "DB302-5.0-2P-GN-S",
    "reelPrice": 0,
    "stockNumber": 12525,
    "productIntroEn": " P=5mm  Screw terminal ROHS",
    "productUnit": "Piece",
    "isReel": false,
    "parentCatalogName": "Connectors",
    "title": "DORABO DB302-5.0-2P-GN-S",
    "isHasBattery": false,
    "overseasStockVO": {
      "total": 12525,
      "ship3Days": 4925,
      "shipImmediately": 7600
    },
    "pdfUrl": "https://datasheet.lcsc.com/lcsc/2301061609_DORABO-DB302-5-0-2P-GN-S_C2997275.pdf",
    "wmStockHk": 0,
    "catalogId": 11140,
    "split": 5,
    "stockJs": 4925,
    "domesticStockVO": {
      "total": 12525,
      "ship3Days": 4925,
      "shipImmediately": 7600
    },
    "productPriceList": [
      {
        "currencyPrice": 0.2077,
        "discountRate": "0.80",
        "ladder": 5,
        "usdPrice": 0.2077,
        "currencySymbol": "$",
        "productPrice": "0.2077"
      },
      {
[... 269 lines cut ...]
        "isMain": true,
        "sortNumber": 1,
        "paramNameEn": "LED",
        "paramName": "LED",
        "paramValue": "不带LED"
      },
      {
        "paramValueEn": "-40℃~+85℃",
        "paramCode": "param_13480_s",
        "isMain": true,
        "sortNumber": 1,
        "paramNameEn": "Operating Temperature Range",
        "paramName": "工作温度范围",
        "paramValue": "-40℃~+85℃"
      }
    ],
    "parentCatalogId": 365,
    "brandId": 11722,
    "brandNameEn": "XKB Connection",
    "maxBuyNumber": -1,
    "isHot": false
  }
}
```
</details>

T0jan, 2024-06-12:
fixed both in pull request #236. Second one was caused by a too long part name, Inventree allows only 100 characters in the part name, there is now an error message to report this specific error in the gui.

NightSkySK, 2024-06-13:
@T0jan thank you, the implemented solution looks solid. Let's hope, it will be merged to main and release soon.

eeintech, 2024-06-14:
Fixed in #236

closed 2024-06-14 by eeintech