sparkmicro/Ki-nTree #235: Creating parts/stock items in inventree from Ki-nTree creates part with off by one quantity

timmyhadwen, 2024-06-06:
What i did:
- Creating parts by entering in Part Number in search field
- Jumping to InvenTree field to select the right category, quantity and stock location
- Not using KiCad plugin

What happens:
- Each part goes into inventree stock as the quantity of the previous entry.

Create part A with quantity 1 - creates part A with quantity 1
Create part B with quantity 2 - creates part B with quantity 1
Create part C with quantity 3 - creates part C with quantity 2
etc...

eeintech, 2024-06-13:
@timmyhadwen Ummm will need to reproduce and fix this, thanks for reporting it!

T0jan, 2024-06-14:
I can confirm this on Windows and Linux. There seems to be a problem with updating the value while the user is entering it. If you enter a value, switch to the `Create` view and create a part the stock number will be set to whatever was set before in the `Stock Quantity`. If you enter a value, switch to any other view, switch back to the `InvenTree` view and then create the part afterwards the correct quantity is set. It seems like the corresponding background variable is only updated when entering the view, not while exiting or changing the value.

T0jan, 2024-06-14:
And it was as expected, some callbacks were missing, fix is ready: #237

T0jan, 2024-07-08:
fixed in v1.1.1

closed 2024-07-08 by T0jan

timmyhadwen, 2024-09-04:
<3