sparkmicro/Ki-nTree #238: Parsing error in parameters_filters.yaml causes file to be over-written

hillsandales, 2024-06-19:
Given this custom file:

**parameters_filters.yaml**
```
Capacitors:
- Value
- Rated Voltage
- Tolerance
- Package Type
- Temperature Grade
Circuit Protections:
- Value
Connectors:
- Value
Crystals and Oscillators:
- Value
- Package Type
- Package Size
- Temperature Range
Diodes:
- Value
Inductors:
- Value
- Rated Current
- Package Type
- Package Size
- Temperature Range
Integrated Circuits:
- Value
Mechanicals:
- Value
Modules:
- null
Power Management:
- Value
Printed-Circuit Boards:
- null
RF:
- Value
Resistors:
- Value
- Tolerance
- Rated Power
- Package Type
- Temperature Range
Transistors:
- Value
- Collector Gate Voltage
- ECCN
- Footprint
- Frequency
- Function Type
- Gain (hFE)
- Maximum Gate Voltage
- Mounting Type
- Package Type
- Packaging
- Rated Current
- Rated Power
- Rated Voltage
- RDS On Resistance
- Symbol
- Temperature Range
```

If there's a formatting error on the first item in the list under a category, it causes the entire file to be overwritten by the default file.

If by accident, the file changes like this for one category:

```
Transistors:
   - Value
- Collector Gate Voltage
- ECCN
- Footprint
- Frequency
- Function Type
- Gain (hFE)
- Maximum Gate Voltage
- Mounting Type
- Package Type
- Packaging
- Rated Current
- Rated Power
- Rated Voltage
- RDS On Resistance
- Symbol
- Temperature Range
```
When the program is re-launched, and the file is checked, the file is over-written and looks like this:
```
Capacitors:
- Value
- Rated Voltage
- Tolerance
- Package Type
- Temperature Grade
Circuit Protections:
- Value
Connectors:
- Value
Crystals and Oscillators:
- Value
- Package Type
- Package Size
- Temperature Range
Diodes:
- Value
Inductors:
- Value
- Rated Current
- Package Type
- Package Size
- Temperature Range
Integrated Circuits:
- Value
Mechanicals:
- Value
Modules:
- null
Power Management:
- Value
Printed-Circuit Boards:
- null
RF:
- Value
Resistors:
- Value
- Tolerance
- Rated Power
- Package Type
- Temperature Range
Transistors:
- Value
```
All custom settings are lost. I ran this same test, and added a parameter under a separate category as well. If an error shows up, that other category was also wiped

If an item further down the list is indented by accident like:
```
Transistors:
- Value
- Collector Gate Voltage
- ECCN
- Footprint
- Frequency
- Function Type
- Gain (hFE)
- Maximum Gate Voltage
- Mounting Type
- Package Type
   - Packaging
- Rated Current
- Rated Power
- Rated Voltage
- RDS On Resistance
- Symbol
- Temperature Range
```
The new file when relaunched looks like this:
```
Transistors:
- Value
- Collector Gate Voltage
- ECCN
- Footprint
- Frequency
- Function Type
- Gain (hFE)
- Maximum Gate Voltage
- Mounting Type
- Package Type - Packaging
- Rated Current
- Rated Power
- Rated Voltage
- RDS On Resistance
- Symbol
- Temperature Range
```
The item is moved up to the previous row, but the rest of the file is not over-written.

The terminal reported things like this:
```
$ poetry run python -m kintree_gui
while parsing a block mapping
  in "C:\Users\user\kintree\user\parameters_filters.yaml", line 1, column 1
expected <block end>, but found '-'
  in "C:\Users\user\kintree\user\parameters_filters.yaml", line 44, column 1
```
 So it appears to know where in the file something is formatted incorrectly.

Question: Why does the program automatically rewrite or reformat these config files? Is that necessary? Could it instead catch a format exception for the YAML and just exit before launching? If it doesn't launch due to a formatting error, it would allow the formatting error to be fixed and not lose custom settings.

I was changing a lot of these config files trying to get my parameters to populate when creating a part (which is proving difficult as the terminal reports they were not added but I can't figure out why from the output, but not related to this bug), and lost all my custom configured settings. :-(

T0jan, 2024-06-21:
Rn there is one single function to load all different kinds of config files, resulting in overwriting the user file with the template if anything goes wrong during the load. Due to #228 the handling will most likely change in the future, probably resulting in a slightly different handling of program settings and parameter settings, solving this aspect here too.

I am not sure what would be the best way here instead of overwriting, either starting the program with an visible error message but still being able to create new parts without the functionality or just an error and closing the program right away. I tend to the first one.

eeintech, 2024-06-21:
@hillsandales Thanks for reporting, it does sound like a niche case scenario and hadn't really ran into it before. Sorry for the data loss... hopefully you got this going after all your effort. It is a sure feature that needs a bit more polishing and I would appreciate any help improving it 😃