sparkmicro/Ki-nTree #239: Part parameters not being added at part creation

hillsandales, 2024-06-19:
I'm adding some parts using the Digi-Key supplier option. I have parameters mapped and I'm finding that the parameters are not being added when the part is being created. However, once the part is created, and I run an Update on the IPN to update parameters, the parameters are added.

Here's the output from a part I ran:
```
[MAIN]  Generating Internal Part Number
[INFO]  Internal Part Number = TRA-0091-0000
[INFO]  Success: Added new part to InvenTree
[INFO]  Warning: Image download failed (HTTP Error)

[MAIN]  Creating parameters
[TREE]  Error: Parameter template "Technology" does not exist
[TREE]  Warning: The following parameters were skipped:
--->    Collector Gate Voltage
--->    ECCN
--->    Frequency
--->    Function Type
--->    Gain (hFE)
--->    Mounting Type
--->    Package Type
--->    Packaging
--->    Rated Current
--->    Rated Power
--->    Rated Voltage
--->    Temperature Range
--->    Value
--->    Maximum Gate Voltage
--->    RDS On Resistance
--->    Technology

[MAIN]  Creating manufacturer part
[INFO]  Success: Added new manufacturer part

[MAIN]  Creating supplier part
[INFO]  Success: Added new supplier part

[MAIN]  Processing Price Breaks
[INFO]  Success: The price breaks were updated

[MAIN]  Opening URL https://[REMOVED]/part/TRA-0091-0000/ in browser
[TREE]  Successfully connected to InvenTree server (ENV=DEVELOPMENT)

[MAIN]  Searching for original part in database
[INFO] Success: Found original part in database (ID = 91 | Description = "BJT TO92 80V NPN 0.625W 150C")
[INFO]  Warning: The following parameters were not found in supplier data:
['Vgs (Max)', 'Rds On (Max) @ Id, Vgs', 'Technology']
[INFO]  The following parameters are not mapped in Digi-Key parameters configuration:
['Package / Case']

[MAIN]  Creating parameters
[TREE]  Error: Parameter template "Technology" does not exist
[INFO]  Success: The following parameters were updated:
--->    Collector Gate Voltage
--->    ECCN
--->    Frequency
--->    Function Type
--->    Gain (hFE)
--->    Mounting Type
--->    Package Type
--->    Packaging
--->    Rated Current
--->    Rated Power
--->    Rated Voltage
--->    Temperature Range
--->    Value
[TREE]  Warning: The following parameters were skipped:
--->    Maximum Gate Voltage
--->    RDS On Resistance
--->    Technology

[MAIN]  Creating manufacturer part
[INFO]  Manufacturer part already exists, skipping.

[MAIN]  Creating supplier part
[INFO]  Supplier part already exists, skipping.

[MAIN]  Processing Price Breaks
[INFO]  Success: The price breaks were updated
```

Nothing else was done between creating the part, and Updating the part params.

I'm running Ki-nTree version 1.0.0-dev (from what it says in GUI). I downloaded the latest from Github and installed it with the Poetry instructions. Somehow I had installed it with pip, and I got a version that said it was version 1.1.0, but I can't seem to figure out how to get it again from Github.

InvenTree version is 0.15.3
API version 196
Python version 3.11.9
Django version 4.2.12

T0jan, 2024-06-21:
The main issue here is that you have the Parameter `Technology` mapped in your `supplier_parameters.yaml` but you don't have a matching parameter template in your inventree database, so this parameter gets skipped as intended.

The other parameters, at least the ones which are in the supplier data, seem to be created as expected, because your second run does not report them being created but instead being updated, so they were already created in the first run. For some reason I do not understand rn they get sorted into the wrong list after the creation and therefore get reported as skipped instead of created.

eeintech, 2024-06-21:
Using the repo configuration, I can't reproduce:

``` shell
[MAIN]  Digi-Key search for MPSA06
[TREE]  Successfully connected to InvenTree server (ENV=DEVELOPMENT)
[INFO]  Warning: The following parameters were not found in supplier data:
['Vgs (Max)', 'Rds On (Max) @ Id, Vgs']
[INFO]  The following parameters are not mapped in Digi-Key parameters configuration:
['Packaging', 'Package / Case', 'Mounting Type', 'Current - Collector Cutoff (Max)', 'DC Current Gain (hFE) (Min) @ Ic, Vce', 'Frequency - Transition', 'ECCN']

[MAIN]  Generating Internal Part Number
[INFO]  Internal Part Number = TRA-000059-00
[INFO]  Success: Added new part to InvenTree
[INFO]  Warning: Image download failed (HTTP Error)

[MAIN]  Creating parameters
[TREE]  Error: Parameter template "Collector-Gate Voltage" does not exist
[INFO]  Success: The following parameters were created:
--->    Function Type
--->    Package Type
--->    Rated Current
--->    Rated Power
--->    Rated Voltage
--->    Temperature Range
--->    Value
--->    Maximum Gate Voltage
--->    RDS On Resistance
[TREE]  Warning: The following parameters were skipped:
--->    Collector-Gate Voltage
```

@hillsandales Could you please share your `supplier_parameters.yaml`? If you have made changes there maybe that'll help us figure this one out.

hillsandales, 2024-06-23:
Here's my **supplier_parameters.yaml** file. I had added a lot of parameters and changed around categories in Inventree from the default config. Maybe I should have kept it simple...

I'm a little unclear on what each config file does. categories.yaml makes sense and the supplier_parameters.yaml is starting to make sense. For the supplier categories, I thought each supplier had to have its settings put into a separate file like digikey was but then I read through the docs carefully and found a sentence or something that noted only Digikey works for a file like digikey_categories.yaml.

I've been developing a fork of Ki-nTree to add another supplier's "API" (not really, more like LCSC's method) so I've been through the source code a lot more now than a week ago and trying to wrap my head around all the  config files.

I'm still unclear on what each setting does in the <supplier>_config.yaml file, such as EXTRA_FIELDS, SEARCH_DATASHEET, etc. Do these SEARCH_<field> settings allow a user to specify a different parameter to get the value for, say, the NAME field? e.g. If I wanted the NAME field to be filled in with the Manufacturer_Part_Number, how would I do that with the <supplier>_config.yaml file settings?

```
Base:
  Number of Contacts:
  - Number of Terminations
  Packaging:
  - Packaging
  Temperature Range:
  - Operating Temperature
Capacitors:
  ESR:
  - ESR (Equivalent Series Resistance)
  Package Height:
  - Height - Seated (Max)
  - Thickness (Max)
  Package Size:
  - Size / Dimension
  Rated Voltage:
  - Voltage - Rated
  - Voltage Rated
  Temperature Grade:
  - Temperature Coefficient
  Temperature Range:
  - Operating Temperature
  Tolerance:
  - Tolerance
  Value:
  - Capacitance
  parent:
  - Base
Circuit Protections:
  Breakdown Voltage:
  - Voltage - Breakdown (Min)
  Capacitance:
  - Capacitance @ Frequency
  Clamping Voltage:
  - Voltage - Clamping (Max) @ Ipp
  Rated Current:
  - Current Rating (Amps)
  - Current - Max
  Rated Power:
  - Power - Peak Pulse
  Rated Voltage:
  - Voltage Rating - DC
  - Voltage - Max
  Standoff Voltage:
  - Voltage - Reverse Standoff (Typ)
  Value:
  - Manufacturer Part Number
  parent:
  - Base
Connectors:
  Features:
  - Features
  Frequency:
  - Frequency - Max
  Interface Type:
  - Connector Type
  - Flat Flex Type
  Locking:
  - Locking Feature
  - Fastening Type
  Mating Height:
[... 232 lines cut ...]
  Mounting Type:
  - Mounting Type
  Package Type:
  - Supplier Device Package
  Packaging:
  - Packaging
  RDS On Resistance:
  - Rds On (Max) @ Id, Vgs
  Rated Current:
  - Current - Collector (Ic) (Max)
  - "Current - Continuous Drain (Id) @ 25\xB0C"
  - Current - Collector Cutoff (Max)
  Rated Power:
  - Power - Max
  - Power Dissipation (Max)
  Rated Voltage:
  - Voltage - Collector Emitter Breakdown (Max)
  - Drain to Source Voltage (Vdss)
  Technology:
  - Technology
  Temperature Range:
  - Operating Temperature
  Value:
  - Manufacturer Part Number
```

T0jan, 2024-06-24:
@hillsandales please check if you have the `Technology` part parameter template in InvenTree and if not either add it or remove the parameter from your transistor configuration in the `supplier_parameters.yaml`. This should at least solve the cause for your error (not for the bug reporting the other parameters wrongly as not created)

hillsandales, 2024-06-25:
Adding "Technology" as a parameter in Inventree fixed that problem. Parameters still need to be written as a second action after the part is initially created it seems.

T0jan, 2024-06-25:
@hillsandales have you checked if the parameters are there in Inventree after the first run already and are only wrongly shown as skipped?

hillsandales, 2024-06-29:
@T0jan I setup a fresh setup at work, and started populating params from scratch. It does seem like the parameters are not written when the part is created. The terminal shows this but I checked the InvenTree parameters on the website, and they're not populated. Running it again and Updating the part with the part number writes the parameters, both indicating in the terminal and also appears on the Inventree site.

I'm starting to wrap my head around all the config files and how they setup what is sent to InvenTree, so it's possible I still have something mis-configured, but this fresh setup is much cleaner and I'm adding categories, parameters, and such slowly and trying to check things as I add them.

eeintech, 2024-08-07:
Hello @hillsandales I tried again and failed to find any issue with parameters. The `setup_inventree` script should take care of creating templates while the GUI should be able to add them to each created parts on the first try.

If you get an error like:
``` shell
[MAIN]  Creating parameters
[TREE]  Error: Parameter template "(Min) Output Voltage" does not exist
[TREE]  Error: Failed to create part parameter "Frequency".
[TREE]  Error: Parameter value "Up to 2MHz" is not allowed by server settings.
```

This either means:
1. the "(Min) Output Voltage" template is not in your `parameters.yaml`, you need to add it there and re-run the `setup_inventree` script to create the template
2. the `supplier_parameters.yaml` file maps to an incorrect parameter template (here "(Min) Output Voltage" name was wrong as the template name is "Min Output Voltage", no parenthesis).

If you do get an error like:
``` shell
[MAIN]  Creating parameters
[TREE]  Error: Failed to create part parameter "Frequency".
[TREE]  Error: Parameter value "Up to 2MHz" is not allowed by server settings.
```

You'll need to edit the parameter value prior to add the part in InvenTree, from: 
[image]

to:
[image]

Then the output should show:
``` shell
[MAIN]  Creating parameters
[INFO]  Success: The following parameters were created:
--->    Frequency
```

And the parameter be added to your part in InvenTree:
[image]

I will close this ticket for now, if you still have difficulties, please re-open it and copy the Ki-nTree log here.

closed 2024-08-07 by eeintech