sparkmicro/Ki-nTree #241: Problem to create parameter templates in Inventree

vutaf, 2024-06-23:
Hi,

I ran into a problem during the setup:
the Inventree server is new installed, based on the current docker image, nothing customized.
kintree is installed on a Win10 machine via pip, version is 1.1.0
```
PS C:\Users\XXXX> python --version
Python 3.11.9
```

when I start kintree.setup_inventree, I ran into following problem:
(I added additional print in setup_inventree.py (line 46) to get the list of parameters at start, and in inventree_api.py to get the parameters of create_parameter_template() (line 825))
Also set SETUP_CATEGORIES to false, as it worked without problems

```
PS C:\Users\XXXX> python -m kintree.setup_inventree

[MAIN]  Starting InvenTree setup
[MAIN]  Connecting to Inventree
[TREE]  Successfully connected to InvenTree server (ENV=DEVELOPMENT)
{
    "Antenna Type": null,
    "B Constant": "K",
    "Breakdown Voltage": "V",
    "Capacitance": "nF",
    "Clamping Voltage": "V",
    "Collector Gate Voltage": "V",
    "DC Resistance": "m\u00ce\u00a9",
    "ESR": "m\u00ce\u00a9",
    "Footprint": null,
    "Forward Voltage": "V",
    "Frequency": "Hz",
    "Frequency Stability": "ppm",
    "Frequency Tolerance": "ppm",
    "Function Type": null,
    "Interface Type": null,
    "LED Color": null,
    "Load Capacitance": "pF",
    "Locking": null,
    "Mating Height": "mm",
    "Max Input Voltage": "V",
    "Max Output Voltage": "V",
    "Maximum Gate Voltage": "V",
    "Memory Size": null,
    "Min Input Voltage": "V",
    "Min Output Voltage": "V",
    "Mounting Type": null,
    "Number of Channels": null,
    "Number of Contacts": null,
    "Number of Elements": null,
    "Number of Rows": null,
    "Orientation": null,
    "Output Current": "A",
    "Output Type": null,
    "Package Height": "mm",
    "Package Size": "mm",
    "Package Type": null,
    "Pitch": "mm",
    "Polarity": null,
    "Quiescent Current": "A",
    "RDS On Resistance": "\u00ce\u00a9",
    "RDS On Voltage": "V",
    "Rated Current": "A",
    "Rated Power": "W",
    "Rated Voltage": "V",
    "Saturation Current": "A",
    "Shielding": null,
    "Standoff Voltage": "V",
    "Symbol": null,
    "Temperature Grade": null,
    "Temperature Range": "\u00b0C",
    "Tolerance": "%",
    "Value": null
}

[MAIN]  Loading Parameters
Antenna Type None
[TREE]  Warning: Parameter "Antenna Type" already exists
B Constant K
[TREE]  Warning: Parameter "B Constant" already exists
Breakdown Voltage V
[TREE]  Warning: Parameter "Breakdown Voltage" already exists
Capacitance nF
[TREE]  Warning: Parameter "Capacitance" already exists
Clamping Voltage V
[TREE]  Warning: Parameter "Clamping Voltage" already exists
Collector Gate Voltage V
[TREE]  Warning: Parameter "Collector Gate Voltage" already exists
DC Resistance mÎ©
[TREE]  Error: Failed to create parameter template "DC Resistance".
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\python\Lib\site-packages\kintree\setup_inventree.py", line 57, in <module>
    setup_inventree()
  File "C:\python\Lib\site-packages\kintree\setup_inventree.py", line 49, in setup_inventree
    pk = inventree_api.create_parameter_template(name, unit)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\python\Lib\site-packages\kintree\database\inventree_api.py", line 839, in create_parameter_template
    if parameter_template:
       ^^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'parameter_template' where it is not associated with a value
```
For me, it looks like a problem with the encoding, but I'm running out of ideas where to look furter. 
Does someone have a hint?

Thanks a lot

T0jan, 2024-06-24:
yeah, that's an encoding issue. For a quick fix to get your setup up and running go to your local Ki-nTree config folder, open the `user/parameters.yaml` and replace/remove the units with special characters (should be only the ones with Ω). Even better if you could try to replace the file with this one here: [parameters.yaml.txt](https://github.com/user-attachments/files/15953474/parameters.yaml.txt) (remove the .txt from the file name after downloading it) so I could directly test if the quick fix I have in mind works.

vutaf, 2024-06-24:
Just check, with your parameters.yaml the script is working. Thanks a lot :)

T0jan, 2024-06-25:
updated the parameters config file: https://github.com/sparkmicro/Ki-nTree/commit/49e07ae769dad850c7657a38b67fd8c1ff686bd9

closed 2024-06-25 by T0jan