sparkmicro/Ki-nTree #242: Crashes after a certain LCSC SKU

PricelessToolkit, 2024-06-24:
Ki-nTree crashes after a certain SKU "C2757850, C356725"

I don't know if its a bug or only on my system.
Version 1.1.0
Python 3.9

Can someone confirm?
Thank you!

`[MAIN]  LCSC search for C2757850

Future exception was never retrieved
future: <Future finished exception=KeyError('paramVOList')>
Traceback (most recent call last):
  File "C:\Users\dmour\AppData\Local\Programs\Python\Python39\lib\concurrent\futures\thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\dmour\AppData\Local\Programs\Python\Python39\lib\site-packages\flet_core\page.py", line 528, in wrapper
    handler(*args)
  File "C:\Users\dmour\AppData\Local\Programs\Python\Python39\lib\site-packages\kintree\gui\views\main.py", line 330, in run_search
    part_supplier_info = inventree_interface.supplier_search(
  File "C:\Users\dmour\AppData\Local\Programs\Python\Python39\lib\site-packages\kintree\database\inventree_interface.py", line 480, in supplier_search
    part_info = lcsc_api.fetch_part_info(part_number)
  File "C:\Users\dmour\AppData\Local\Programs\Python\Python39\lib\site-packages\kintree\search\lcsc_api.py", line 99, in fetch_part_info
    for parameter in range(len(part[parameter_key])):
KeyError: 'paramVOList'`

T0jan, 2024-06-25:
known and fixed already, will be part of the next release: #234

closed 2024-06-25 by T0jan

eeintech, 2024-06-28:
Release `1.1.1` is ready for download 😃

PricelessToolkit, 2024-06-29:
Nice!!!!!!!!!!!!