sparkmicro/Ki-nTree #243: Official way to run the app as a web server

LukeARESYS, 2024-06-25:
Do you have a recommended method for running the KinTree app as a web server? We would like to have this functionality to be able to run this plugin on the same server where we're hosting Inventree, in this way we can ensure every user can access a correctly configured KinTree instance.

I've tried following the official Flet [self hosting](https://flet.dev/docs/publish/web/dynamic-website/hosting/self-hosting/) guide and I had mixed results: for various reasons I had to install KinTree on three different machines and the same setup worked on two machines only, but making it work was kind of a hassle.

I've installed kintree as sudo using pip, I succesfully setup nginx as described in the flet guide and I've created a systemd unit file like this:

```
[Unit]
Description=Kintree
After=network.target

[Service]
User=xxx
Group=xxx
Environment="FLET_SERVER_PORT=6001"
Environment="FLET_FORCE_WEB_SERVER=true"
Environment="FLET_WEB_APP_PATH=/kintree"
ExecStart=kintree

[Install]
WantedBy=multi-user.target
```

I then had to modify the kintree_gui.py file by changing `def main(view='flet_app'):` to `def main(view='browser'):`.

This is just a rough description of the steps I've followed just to give you an idea. Do you think these steps are recommended or do you suggest an alternative method?

If you need a more precise step-by-step description of the issues I've encountered I could try installing kintree again on a fresh machine.

NightSkySK, 2024-06-28:
It would be awesome if self-hosting Kintree would be possible. I would go into this solution once it is supported. 
I use Kintree by myself, but I use a few computers to work with Kintree, so my short-term solution was setting config files on a mapped network drive. Each instance uses the same configuration, however self-hosting instance would be a much better solution.

eeintech, 2024-06-28:
Hello there, thank you for your interest in Ki-nTree 😃 

Yes it is possible to run the app in the browser but I am not sure about self-hosting it.
The main problem with running it as webapp is KiCad file access. No webapp is allowed to access local files, for security reasons (which is why in the first place I designed as a desktop app, Felt came with version v1 and webapp possibility with it).
If you do not use KiCad, then I don't see why not but I do not have the know-how at this time and very limited bandwidth to dig into it, sorry for the bad news.

However I am open to hear any solution you may find and, if not too much trouble, add it as a feature and/or in the documentation 😃

LukeARESYS, 2024-07-18:
Hi, I can confirm that following the [flet self-hosting guide](https://flet.dev/docs/publish/web/dynamic-website/hosting/self-hosting/) we are succesfully running the kintree app as a webserver since quite a while. However, we are only using kintree for managing an InvenTree server, we are not managing a KiCad library, which as you've stated could add some trouble due to file web access.

Sadly it looks like it is kind of unstable, as after a while kintree seems to crash and restarting the Kintree service via systemctl seems to be necessary. The errors reported by systemctl status are quite random each time, so for now we were unable to find a precise.

> I then had to modify the kintree_gui.py file by changing def main(view='flet_app'): to def main(view='browser'):.

Modifying the kintree source code is actually not necessary, the `FLET_FORCE_WEB_SERVER=true` environment variable is enough.

Of course, disabling the Kintree user setting `Open Browser after Creating Part` is necessary, otherwise Kintree will crash.

eeintech, 2024-07-19:
Thanks for the report @LukeARESYS. It's great to know Ki-nTree can run in the browser, although unstable. Do you have multiple people use it simultaneously?