sparkmicro/Ki-nTree #244: Error when trying to download image and datasheet

martin323232, 2024-06-26:
Hello,
when image and/or datasheet url is present on part create it fails. 
From CMD on windows below.

> ←[95m
[MAIN]  Digi-Key search for 1734-1838-ND←[0m
←[92m[TREE]     Successfully connected to InvenTree server (ENV=DEVELOPMENT)←[0m
←[93m[INFO]     Warning: Parameter map for "Tools and Production Supplies" does not exist or is empty←[0m
←[95m
[MAIN]  Generating Internal Part Number←[0m
[INFO]  Internal Part Number = TOL-000630-00
←[92m[INFO]     Success: Added new part to InvenTree←[0m
←[93m[INFO]     Warning: Image download failed (HTTP Error)←[0m
Future exception was never retrieved
future: <Future finished exception=OSError(22, 'Invalid argument')>
Traceback (most recent call last):
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python311\Lib\concurrent\futures\thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python311\Lib\site-packages\flet_core\page.py", line 528, in wrapper
    handler(*args)
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\gui\views\main.py", line 1442, in create_part
    new_part, part_pk, part_info = inventree_interface.inventree_create(
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\database\inventree_interface.py", line 651, in inventree_create
    datasheet_link = inventree_api.upload_part_datasheet(inventree_part['datasheet'], part_pk)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\database\inventree_api.py", line 477, in upload_part_datasheet
    if not download_with_retry(datasheet_url,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\common\tools.py", line 124, in download_with_retry
    file = download(url, fileoutput=full_path, silent=silent, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\common\tools.py", line 93, in download
    (file, headers) = urllib.request.urlretrieve(url, filename=fileoutput)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python311\Lib\urllib\request.py", line 251, in urlretrieve
    tfp = open(filename, 'wb')
          ^^^^^^^^^^^^^^^^^^^^

martin323232, 2024-06-26:
EDIT: Works if I disable Datasheet upload

closed 2024-06-26 by T0jan

eeintech, 2024-06-28:
Release `1.1.1` is ready for download 😃