sparkmicro/Ki-nTree #249: Issue installing on Windows /  DLL load failed

ashleyvansteenacker, 2024-06-30:
Hi I seem to have some issues to launch Ki-nTree. 
I tried it on another computer and had the same issue.
I'm not a Python expert, So I'm probably looking over something stupid.

Followed the installation process on the Git homepage

`PS C:\Users\Ashley\Ki-nTree> kintree
Traceback (most recent call last):
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Ashley\AppData\Local\Programs\Python\Python39\Scripts\kintree.exe\__main__.py", line 4, in <module>
    from kintree.kintree_gui import main
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\kintree\kintree_gui.py", line 3, in <module>
    from .gui.gui import kintree_gui
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\kintree\gui\gui.py", line 4, in <module>
    from .views.main import (
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\kintree\gui\views\main.py", line 19, in <module>
    from ...database import inventree_interface
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\kintree\database\inventree_interface.py", line 8, in <module>
    from ..search import search_api, digikey_api, mouser_api, element14_api, lcsc_api, tme_api
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\kintree\search\digikey_api.py", line 3, in <module>
    import digikey
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\digikey\__init__.py", line 1, in <module>
    from digikey.v3.api import (keyword_search, product_details, digi_reel_pricing, suggested_parts,
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\digikey\v3\api.py", line 4, in <module>
    import digikey.oauth.oauth2
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\digikey\oauth\oauth2.py", line 14, in <module>
    from certauth.certauth import CertificateAuthority
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\certauth\certauth.py", line 6, in <module>
    from OpenSSL import crypto
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\OpenSSL\__init__.py", line 8, in <module>
    from OpenSSL import SSL, crypto
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\OpenSSL\SSL.py", line 10, in <module>
    from OpenSSL._util import (
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\OpenSSL\_util.py", line 6, in <module>
    from cryptography.hazmat.bindings.openssl.binding import Binding
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\cryptography\hazmat\bindings\openssl\binding.py", line 15, in <module>
    from cryptography.exceptions import InternalError
  File "c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\cryptography\exceptions.py", line 9, in <module>
    from cryptography.hazmat.bindings._rust import exceptions as rust_exceptions
ImportError: DLL load failed while importing _rust: The specified procedure could not be found.`

Pip Version:
`PS C:\Users\Ashley\Ki-nTree> pip --version
pip 24.1.1 from c:\users\ashley\appdata\local\programs\python\python39\lib\site-packages\pip (python 3.9)`

OS: Win 10 64bit

eeintech, 2024-07-02:
Hello @ashleyvansteenacker Python 3.9 should work, do you know which version of Ki-nTree you tried to install?

ashleyvansteenacker, 2024-07-02:
Hey thank you for the reply @eeintech ! I believe it was the latest version 1.1.1

eeintech, 2024-07-02:
@ashleyvansteenacker Really bizarre, I can run Ki-nTree 1.1.1 no problem using PowerShell and Python 3.11. Would you mind trying the same Python version?

ashleyvansteenacker, 2024-07-02:
@eeintech Updating to Python 3.11 did the trick. It seems to be working now!
Thank you!

[image]

closed 2024-07-02 by ashleyvansteenacker

eeintech, 2024-07-03:
@ashleyvansteenacker Great to hear, not sure exactly why 3.9 is not working...