sparkmicro/Ki-nTree #252: Problem uploading the datasheet

simoneamadori, 2024-07-17:
With the last version, when uploading the datasheet, it will upload a file with just ".pdf" without the correct name.
And also it will be always the same file for different components.
[image]

Thanks

eeintech, 2024-07-17:
Sounds like a bug... @T0jan or I should investigate. Thanks for the report. Could you confirm this does not happen with `1.1.1`?

simoneamadori, 2024-07-17:
Thanks for your great job!
I confirm that it doesn't happens in the 1.1.1.

eeintech, 2024-07-17:
Thanks for the confirmation, it rules out the setup as a variable 👍 
It is likely my fault and I will try to fix it soon, really sorry about this!

simoneamadori, 2024-07-18:
don't worry, is a minor trouble.
You guys are doing a great work!

T0jan, 2024-07-19:
@eeintech something is really broken. After upgrading to 1.1.2 neither my linux nor my windows setup even tries to upload datasheets and also no images are being uploaded for me. It also gives no error messages indicating something is wrong...

I took a quick look to the changes since my last commit where I know it was still working but found no obvious errors.

simoneamadori, 2024-07-19:
For the images, I saw that if you do an update of the part the image will be uploaded.

T0jan, 2024-07-19:
good hint! An update of the part uploaded the image as well as the datasheet for me. We have to take a look on the differences between the part creation and update for uploads it seems.

T0jan, 2024-07-19:
good hint, there seem to be multiple issues here, as I could confirm the behavior of the unnamed pdf after updating the part. does the PDF appears only after you updated as well or also for new parts? 

The uploaded PDF seems to be always the first one you uploaded after the update to 1.1.2 and the datasheets of no component after are being downloaded. The first datasheet is stored as `.pdf` in the cache and probably all further runs check, if this specific "file-name" is already existent in the cache and skip the download afterwards.

so the main issues are:
* no downloads or uploads happen on the initial part creation
* on part update the file naming of the PDF fails, causing all files to have no name at all

eeintech, 2024-07-19:
Crap, sorry for the trouble and investigation... I'll do what I can to fix it soon.

eeintech, 2024-08-07:
@simoneamadori I had broke the datasheet upload in `1.1.2`, please try `1.1.3` 😃

closed 2024-08-07 by eeintech

simoneamadori, 2024-08-08:
Perfect, is working correctly.
Thanks for your work!

commit 26407851 (2024-08-07): Fix #252