sparkmicro/Ki-nTree #255: Failing in updating part

simoneamadori, 2024-09-11:
While using the update existing function it gives me this error :

`←[92m[TREE]     Successfully connected to InvenTree server (ENV=DEVELOPMENT)←[0m
←[95m
[MAIN]  Searching for original part in database←[0m
←[92m[INFO] Success: Found original part in database (ID = 219 | Description = "FUSE BOARD MOUNT 62MA 63VDC 0603")←[0m
←[93m[INFO]     Warning: The following parameters were not found in supplier data:
['Package / Case', 'Operating Temperature', 'Voltage - Breakdown (Min)', 'Capacitance @ Frequency', 'Voltage - Clamping (Max) @ Ipp', 'Current Rating (Amps)', 'Current - Max', 'Power - Peak Pulse', 'Voltage Rating - DC', 'Voltage - Max', 'Voltage - Reverse Standoff (Typ)']←[0m
[INFO]  The following parameters are not mapped in Mouser parameters configuration:
['Packaging', 'Standard Pack Qty']
←[95m
[MAIN]  Creating parameters←[0m
←[93m[TREE]     Warning: The following parameters were skipped:←[0m
--->    Value
--->    Package Type
--->    Temperature Range
--->    Breakdown Voltage
--->    Capacitance
--->    Clamping Voltage
--->    Rated Current
--->    Rated Power
--->    Rated Voltage
--->    Standoff Voltage
Future exception was never retrieved
future: <Future finished exception=HTTPError({'detail': 'Error occurred during API request', 'url': 'http://192.168.10.18:1337/api/part/attachment/', 'method': 'GET', 'status_code': 404, 'body': '{"detail": "API endpoint not found", "url": "http://192.168.10.18:1337/api/part/attachment/?part=219"}', 'headers': {'AUTHORIZATION': 'Token inv-cc1fb180f2cebf58c4fb8e027ad99a140bea8d40-20240528'}, 'params': {'part': 219}})>
Traceback (most recent call last):
  File "C:\Users\simone.amadori\AppData\Local\Programs\Python\Python311\Lib\concurrent\futures\thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\simone.amadori\AppData\Local\Programs\Python\Python311\Lib\site-packages\flet_core\page.py", line 941, in wrapper
    handler(*args)
  File "C:\Users\simone.amadori\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\gui\views\main.py", line 1400, in create_part
    alt_result = inventree_interface.inventree_create_alternate(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\simone.amadori\AppData\Local\Programs\Python\Python311\Lib\site-packages\kintree\database\inventree_interface.py", line 864, in inventree_create_alternate
    attachment = part.getAttachments()
                 ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\simone.amadori\AppData\Local\Programs\Python\Python311\Lib\site-packages\inventree\base.py", line 496, in getAttachments
    return AttachmentSubClass.list(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\simone.amadori\AppData\Local\Programs\Python\Python311\Lib\site-packages\inventree\base.py", line 228, in list
    response = api.get(url=url, params=kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\simone.amadori\AppData\Local\Programs\Python\Python311\Lib\site-packages\inventree\api.py", line 530, in get
    response = self.request(url, method='get', **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\simone.amadori\AppData\Local\Programs\Python\Python311\Lib\site-packages\inventree\api.py", line 378, in request
    raise requests.exceptions.HTTPError(detail)
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url': 'http://192.168.10.18:1337/api/part/attachment/', 'method': 'GET', 'status_code': 404, 'body': '{"detail": "API endpoint not found", "url": "http://192.168.10.18:1337/api/part/attachment/?part=219"}', 'headers': {'AUTHORIZATION': 'Token inv-cc1fb180f2cebf58c4fb8e027ad99a140bea8d40-20240528'}, 'params': {'part': 219}}`

# System & software version:
Version Information:
InvenTree-Version: 0.16.1
Django Version: 4.2.15
Commit Hash: 4c15e5c
Commit Date: 2024-08-30

Database: postgresql
Debug-Mode: False
Deployed using Docker: True
Platform: Linux-5.15.0-119-generic-x86_64-with
Installer: DOC

Active plugins: [{'name': 'InvenTreeBarcode', 'slug': 'inventreebarcode', 'version': '2.1.0'}, {'name': 'InvenTreeCoreNotificationsPlugin', 'slug': 'inventreecorenotificationsplugin', 'version': '1.0.0'}, {'name': 'InvenTreeCurrencyExchange', 'slug': 'inventreecurrencyexchange', 'version': '1.0.0'}, {'name': 'InvenTreeLabel', 'slug': 'inventreelabel', 'version': '1.1.0'}, {'name': 'InvenTreeLabelMachine', 'slug': 'inventreelabelmachine', 'version': '1.0.0'}, {'name': 'InvenTreeLabelSheet', 'slug': 'inventreelabelsheet', 'version': '1.0.0'}, {'name': 'DigiKeyPlugin', 'slug': 'digikeyplugin', 'version': '1.0.0'}, {'name': 'LCSCPlugin', 'slug': 'lcscplugin', 'version': '1.0.0'}, {'name': 'MouserPlugin', 'slug': 'mouserplugin', 'version': '1.0.0'}, {'name': 'TMEPlugin', 'slug': 'tmeplugin', 'version': '1.0.0'}, {'name': 'KiCadLibraryPlugin', 'slug': 'kicad-library-plugin', 'version': '1.4.3'}]

Ki-nTree version 1.1.3

simoneamadori, 2024-09-17:
Resolved updating the inventree package from 0.14.0 to 0.17.0

closed 2024-09-17 by simoneamadori

eeintech, 2024-09-21:
@simoneamadori Correct, I do need to update the inventree distrib I'm using locally to be able to test new InvenTree versions, InvenTree is not keeping up😕