sparkmicro/Ki-nTree #258: AttributeError: 'Page' object has no attribute 'window'

martonmiklos, 2024-09-26:
I tried to run the current main branch and I got:
[image]

I am running KDE Neon (technically it is a 22.04 Ubuntu), all requirements were installed with pip -r
Flet version is 0.22.1

Any advice would be recommended!

T0jan, 2024-09-30:
Could you please add which version of Ki-nTree you are using and any output which is generated in the (python) console you are running Ki-nTree from?

eeintech, 2024-10-01:
@martonmiklos Flet version is incorrect, how do you install Ki-nTree?

Flet version is now at `0.24.1`: `flet = "^0.24.1"`

martonmiklos, 2024-10-01:
@eeintech 
I installed the reqiurements with pip install -r requirements.txt

In the current main the Flet is set to:
`Flet>=0.22.0,<0.23.0`
https://github.com/sparkmicro/Ki-nTree/blob/main/requirements.txt#L3C1-L3C21

Manually altering the requirements and installing Flet 0.24.1 get the thing working.

Many thanks for the quick support!

Shall I file a PR on the requirements.txt?

eeintech, 2024-10-07:
@martonmiklos Good point, I will just keep this ticket opened until `requirements.txt` file is fixed, thanks!